arduino/arduino-pro-ide #195: STM32 boards custom options are not available

TzOk83, 2020-01-31:
The STM32Duino package boards are impossible to select. No options available for them.

ubidefeo, 2020-02-01:
hi @TzOk83 
you're going to have to add the `additional_url` to the CLI configuration file.
It can be a bit tricky but it's on the list to make that easily accessible via UI.

what platform are you on?
if you can find `arduino-cli.yaml` on your system you can add the custom core URL and the boards will show up in Boards Manager

You can see my config file below
```yaml
arduino_data: /Users/xxx/Library/arduino15
board_manager:
  additional_urls:
  - https://dl.espressif.com/dl/package_esp32_index.json
  - http://arduino.esp8266.com/stable/package_esp8266com_index.json
  - https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
daemon:
  port: "50051"
directories:
  data: /Users/xxx/Library/Arduino15
  downloads: /Users/xxx/Library/Arduino15/staging
  user: /xxx/xxx/AppsDocuments/Arduino
logging:
  file: ""
  format: text
  level: info
proxy_type: auto
sketchbook_path: /xxx/xxx/AppsDocuments/Arduino
```
[image]

ornea, 2020-02-01:
I am using Win7 and failed to find arduino-cli.yaml in the user or Arduino_pro directory.  Also used agent-ransack to search for entries shown above such as proxy_type and arduino_data but failed to find a match.  Except embedded in the executable arduino_cli.exe

TzOk83, 2020-02-01:
I'm on windows 7 and Arduino PRO found all boards and libraries installed in Arduino IDE. The problem is I can't access options for the boards.

This is how it looks in Arduino IDE:
[image]

...and in Arduino PRO IDE:
[image]

ubidefeo, 2020-02-01:
@TzOk83 
I understand what you mean, this requires the CLI to drill down into variants and display them in a secondary menu.
the boards are of course in the core, and strangely enough even though they are listed in the `boards.txt` file they are not made available as choices.
the boards of the ESP8266 core are all showing up.

let's wait for our devs to check this issue out, it might just be a parsing bug somewhere :)

per1234, 2020-02-29:
Closing as duplicate of https://github.com/arduino/arduino-pro-ide/issues/10

closed 2020-02-29 by per1234