arduino/arduino-pro-ide #196: Portable Mode is on my wish list.  Would be nice to see that as an option

ornea, 2020-02-01:
**Is your feature request related to a problem? Please describe.**
I am always moving between many PC's and Portable Mode makes this a trivial task.

**Describe the solution you'd like**
Creating a Portable directory in the root folder instructs ArduinoPro to use Portable Mode so all addons, boards and libraries etc are stored within the ArduinoPro directory structure and not buried the User folder
 
**Describe alternatives you've considered**
N/A

**Additional context**
N/A

linxcow, 2020-02-01:
+1 
I use **portable mode** on a thumbdrive with different Ide versions easy to maintain and handy on multiple PC's

keesdekoning, 2020-02-19:
Yes a portable mode and plz not use  ApData and one drive. and store board managers

moelski, 2020-09-27:
Any news on this topic? 
Portable mode is the first thing I enable after downloading any Arduino IDE ...