arduino/arduino-pro-ide #197: Analog sensor doesn't print current value to Serial Monitor

remotesense, 2020-02-01:
**Describe the bug**
I am using an MKR 1010 board and I have a potentiometer on A03
I have varified that my code works fine on normal Arduino IDE serial monitor and directly copied it over into Pro.  It prints the current value of the sensor but does not update it when I adjust the sensor Serial Monitor just stays stuck on the same value.  I need to then re-load the Serial Monitor to get a new value.

**To Reproduce**
Steps to reproduce the behavior:
1. Use a generic script that reads an analog Pin and prints the value to serial monitor
Use an analog sensor like POT that can update manually
2. Click on Serial Monitor, even if its in Autoscroll mode, none of the sensor informaiton updates.

**Expected behavior**
I expected that it would print a new value based on sensor position as I have varified in the old Arduino IDE.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10

owennewo, 2020-02-15:
@remotesense - you forgot the `Serial.begin(9600);` in `void setup()`

remotesense, 2020-02-26:
> @remotesense - you forgot the `Serial.begin(9600);` in `void setup()`
> 
> also I think you want
> `Serial.println(sensorValue);`
> 
> you are printing sensorPin

My Bad, wrong screencap; it was not the code that I tested, the code that I tested did have a baud rate of 9600 and was printing multiple sensor values.  As stated in the comment, it was fully tested and working in normal Arduino IDE... 

Thanks,

Tim

per1234, 2020-03-16:
@remotesense if you change the delay to 100, does the problem go away?

I can reproduce the issue, and also your findings that it does not occur with the original IDE's Serial Monitor. I think what happens is the Arduino Pro IDE's Serial Monitor is too slow to keep up with the data. What you're seeing is it working its way though the backlog of output. So there is just an ever increasing lag before the effects of a change in voltage on `A0` are shown

kittaakos, 2020-07-22:
Was fixed via https://github.com/arduino/arduino-pro-ide/issues/197

closed 2020-07-22 by kittaakos