arduino/arduino-pro-ide #198: Red X next to board and comm port

mikcatta, 2020-02-01:
arduino-pro-ide 0.0.4 on Windows 10 (64bit)

I always get a red X beside the board, even though it works

kittaakos, 2020-02-17:
> I always get a red X beside the board, even though it works

@mikcatta, thanks for reporting the issue. What cores have you installed? What is your selected board and port in the `Select Board` dialog, and what kind of physical board have you attached? Do you see any errors in the console when you start the Pro IDE from a terminal? Thank you!

mikcatta, 2020-02-18:
Hi @kittaakos, Is there any way to tell what core are installed? This is a Seeeduino Mega 2560 connected via USB . The port is COM37. Attached is a copy of the 
console and a screenshot if that is of any assistance? Tried a Sparkfun Pro Mini and I get the same, red X beside the board, but it works.

[console.txt](https://github.com/arduino/arduino-pro-ide/files/4219506/console.txt)
[Screenshot 2020-02-18 13 41 59](https://user-images.githubusercontent.com/3464911/74741366-904e4300-5254-11ea-8a91-d98378cfa2fe.png)

.

kittaakos, 2020-02-18:
Thank you for the update, @mikcatta.

> Is there any way to tell what core are installed?

Yes, from the `Boards Manager` view. You can open it up with <kbd>Ctrl/Cmd</kbd>+<kbd>Shift</kbd>+<kbd>B</kbd>, or from the menu: `Tools` > `Boards Manager...`. All installed cores, have the _INSTALLED_ label. (See the attached image)
[image]

> This is a Seeeduino Mega 2560 connected via USB

I am not a hardware expert, but as far as I see, this is a non-official Arduino boards-package (aka. core). The automatic board detection works only for installed cores, and with the most recent release of the Pro IDE, it is not possible to add a 3rd party, non-official board URL. See #177.

My assumption, based on [this documentation](https://github.com/SeeedDocument/Seeeduino_Mega/blob/master/Seeeduino_Mega.md), is that you have to install the core first, then the board detection will work. But currently, you cannot add a 3rd party boards URL: https://raw.githubusercontent.com/Seeed-Studio/Seeeduino-Boards/master/package_seeeduino_index.json

Note, you can always verify the behavior with the [`arduino-cli`](https://github.com/arduino/arduino-cli#download-the-latest-packages). Below, I wiped all previous state from my local env, attached an Arduino UNO board first, then listed the attached boards (`board list`), then I have attached an Arduino Nano Every board, and listed them (`board list`). If you do the same, and running `board list` prints the name of your board, `Seeeduino Mega 2560`, then it is a bug in the Pro IDE, if no, it is a lack of feature we are working on.

```bash
akos.kitta@Akoss-MacBook-Pro arduino-cli % rm -rf ~/Library/Arduino15 && rm -rf ~/Documents/Arduino/   
akos.kitta@Akoss-MacBook-Pro arduino-cli % ./arduino-cli core update-index
Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded                                                                                                                          
Installing builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 downloaded                                                                                                                       
Installing builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 installed
Updating index: library_index.json downloaded                                                                                                                   
Updating index: package_index.json downloaded                                                                                                                   
akos.kitta@Akoss-MacBook-Pro arduino-cli % ./arduino-cli core list

akos.kitta@Akoss-MacBook-Pro arduino-cli % ./arduino-cli board list
Port                            Type              Board Name  FQBN            Core       
/dev/cu.Bluetooth-Incoming-Port Serial Port       Unknown                                
/dev/cu.MALS                    Serial Port       Unknown                                
/dev/cu.SOC                     Serial Port       Unknown                                
/dev/cu.usbmodem14401           Serial Port (USB) Arduino Uno arduino:avr:uno arduino:avr

akos.kitta@Akoss-MacBook-Pro arduino-cli % ./arduino-cli board list   
Port                            Type              Board Name         FQBN                     Core           
/dev/cu.Bluetooth-Incoming-Port Serial Port       Unknown                                                    
/dev/cu.MALS                    Serial Port       Unknown                                                    
/dev/cu.SOC                     Serial Port       Unknown                                                    
/dev/cu.usbmodem14601           Serial Port (USB) Arduino Nano Every arduino:megaavr:nona4809 arduino:megaavr
```

Finally, can the Java-based Arduino IDE (https://github.com/arduino/Arduino) recognize the attached boards automatically?

mikcatta, 2020-02-18:
Hi @kittaakos 

I have found and attached a Native Arduino Mega and the Red X disappears. It is also listed with "arduino_cli.exe board list". If I try the same, with the Seeeduino Mega2560, I get unknown, so I guess that the is board is currently not supported. 

Port  Type              Board Name FQBN Core
COM37 Serial Port (USB) Unknown

Thanks for your assistance.

kittaakos, 2020-02-18:
Thank you for checking it and reporting back with your findings; much appreciated 🙏 

I am going to reopen it and track this issue _as part of_ #177.

per1234, 2020-03-16:
The Seeduino Mega 2560 uses the FTDI FT232RL USB to TTL serial adapter:
[image]
Since the VID/PID pair of the FT232 doesn't match the VID/PIDs defined in the Mega 2560's boards.txt entry, the Arduino Pro IDE adds this X.

The situation is the same with the Pro Mini. Since that board doesn't have an onboard USB to serial adapter, a general purpose adapter will be used, thus there is no VID/PID defined for this board and no match.

I don't think this behavior is desirable. Although perfect support for a Mega 2560 derivative might not be a high priority, the same issue occurs with the official Arduino Nano, which also uses the FT232 (since it uses the manufacturer's stock VID/PID, there is no VID/PID defined in boards.txt for the Nano, thus there can never be a match).

I would suggest emulating the behavior of the Arduino Web Editor in this respect. In the Arduino Web Editor, when you manually select a port for a board, it does not show the red X in the board selection menu until that port disappears.

kittaakos, 2020-03-16:
> I would suggest emulating the behavior of the Arduino Web Editor in this respect. In the Arduino Web Editor, when you manually select a port for a board, it does not show the red X in the board selection menu until that port disappears.

Nice. Thank you for the suggestion, @per1234. Let's do that.

kittaakos, 2020-03-17:
I have completely rewritten the logic:

Preconditions:
 - I have three boards connected to my laptop:
   - Arduino Uno (Replica),
   - Arduino MKR1000,
   - NodeMCU 0.9 (from the 3rd party ESP8266 core).

Open the app and see the available, attached boards:
[image]

You can select any of the recognized (`arduino-cli board list`) boards:
[image]

If you select the `Unknown` board that is connected to the `/dev/cu.SLAB_USBtoUART` port in my case, the `Bords Config` is opened and the selected port is pre-filled. You have to pick a board:
[image]

Once you have picked a board that was not recognized, we use the name of your selected board:
[image]

kittaakos, 2020-03-17:
Ohh, forgot to mention; since we have guessed the board name and it does not come from the CLI, there is a warning. Instead of the `Red X next to board and comm port`

[image]

closed 2020-05-05 by kittaakos