arduino/arduino-pro-ide #199: Dark mode options

remotesense, 2020-02-01:
In the original Arduino IDE I have a custom theme installed.
It would be nice to see a dark option that has the dark blue-grey tone and nicely contrasted text in a similar style to this one.
See picture as example:
[image]

While its low priority, I find the colors of this theme relaxing on the eyes.
You can find the original theme for reference here.  It is modeled after VSCode theme "One Dark Pro"

https://create.arduino.cc/projecthub/konradhtc/one-dark-arduino-modern-dark-theme-for-arduino-ide-2fca81

kittaakos, 2020-04-02:
Good idea; with the next version we want to support any VS Code themes.

[image]

----

[image]

ubidefeo, 2021-02-19:
brought this one into #58

closed 2021-02-19 by ubidefeo