arduino/arduino-pro-ide #200: programmer is not responding

StipeElektrotechnik, 2020-02-01:
MacOS 
Arduino Nano 
Blink 
don't want upload ?? 
[image]

ubidefeo, 2020-02-01:
there's a red X next to your board, it might not be visible to your system.
try disconnecting/reconnecting it.
does it work with the regular IDE?

StipeElektrotechnik, 2020-02-01:
> there's a red X next to your board, it might not be visible to your system.
> try disconnecting/reconnecting it.
> does it work with the regular IDE?

Yes its works. With the regular IDE

StipeElektrotechnik, 2020-02-01:
I see on the old IDE u can switch the programmer 

I doesn’t finde it on the new IDE ?

ubidefeo, 2020-02-01:
there are functionalities and features which are being implemented and others which can break every now and then. the perks of Alpha versions :D

tomorrow I'll run a test with a Nano as soon as I get to my lab.
which version of Mac OS are you on? I'm on Mojave.
I doubt it's an OS thing but good to have as much information as possible

StipeElektrotechnik, 2020-02-03:
10.15.2

per1234, 2020-03-16:
I've explained the cause of the red X issue here:
https://github.com/arduino/arduino-pro-ide/issues/198#issuecomment-599355836

I suspect the cause of the failed upload is https://github.com/arduino/arduino-pro-ide/issues/10. @StipeElektrotechnik, in the Arduino IDE, what do you have selected in the **Tools > Processor** menu when you are using your Nano?

>I see on the old IDE u can switch the programmer

https://github.com/arduino/arduino-cli/issues/27 is at the heart of this. But that has nothing to do with the upload error you are getting. The Arduino IDE's **Tools > Programmer** menu selection only matters when you are doing an Upload Using Programmer or a Burn Bootloader. It makes no difference when you are doing a normal upload.

closed 2021-02-18 by per1234