arduino/arduino-pro-ide #201: Serial Monitor only shows the incoming data after it receives '\n' character

Master811129, 2020-02-03:
**Describe the bug**
Serial Monitor only shows the incoming data after it receives '\n' character

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Arduino Pro IDE v0.0.4'
2. upload this sketch:
```cpp
void setup()
{
    Serial.begin(9600);
}

void loop()
{
    for (uint8_t a=0; a<12; a++)
    {
        Serial.print("hello world!");
        delay(1000);
    }
    Serial.write('\n');//if you comment this line you CANNOT see anything in serial monitor
}

```
3. comment Serial.write('\n');
4. re-upload the sketch
5. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

kittaakos, 2020-03-02:
Fixed. Thank you for the steps and the snippet, @Master811129!

[image]

closed 2020-03-02 by kittaakos

Master811129, 2020-03-14:
> Fixed. Thank you for the steps and the snippet, @Master811129!
> 
> [image]

thanks man.
when can we see 0.0.6

kittaakos, 2020-03-14:
> thanks man.

Thank you for your detailed bug/feature reports.

> when can we see 0.0.6?

I cannot say it for sure; we target the end of March. The release is planned to be shipped with 3rd part boards package (core) support via the Arduino CLI configuration. Plus we want to include full 3rd party, board-specific core settings (#10.)