arduino/arduino-pro-ide #204: Arduino.h cannot be included in own library

PBahner, 2020-02-06:
With the Arduino Pro IDE I cant include the `<Arduino.h>` in my header and cpp file.
[image]

owennewo, 2020-02-15:
I get the same.  

I believe it's just a linting error - i.e. if your Gleisbildpult.ino includes your Weiche.h and there are no compilation errors in Weiche.h then it will verify/upload. 

I'm also seeing linting errors on the 'standard libraries':
e.g. the servo library described here https://www.arduino.cc/en/Reference/Libraries

I have to ignore the errors and intellisense doesn't work for them either. But I can compile and upload.

MicheleCioccarelli, 2020-03-02:
I have the same issue, but it works fine: it doesn't give any compiling or uploading errors.
it is just visual

spoenemann, 2020-03-04:
#149 must be solved so clangd can find included libraries.

per1234, 2021-02-18:
Hi @PBahner . Thanks so much for taking the time to report this!

This should now be working in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4).

If you try it out and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-18 by per1234