arduino/arduino-pro-ide #205: Bug no autocomplete

Ferreira31, 2020-02-07:
*Version 0.0.4
Muitas coisa ja melhorou no autocomplete, mas algumas funções basicas continuam não aparecendo, por exemplo a Serial 
[image]

owennewo, 2020-02-15:
Are you on windows 10?  Could be the same issue as I had - see #210

per1234, 2021-02-18:
Hi @Ferreira31. Thanks so much for taking the time to report this!

This should now be working in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4).

If you try it out and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-18 by per1234