arduino/arduino-pro-ide #206: "Comment / Uncomment" option

OslafEman, 2020-02-07:
**Is your feature request related to a problem? Please describe.**
Using the original IDE I loved the option "Comment / Uncomment" available with the secondary mouse click.  

**Describe the solution you'd like**
Could be very nice to see it implemented in the new Pro IDE.

owennewo, 2020-02-15:
@OslafEman - just to clarify you want to see that in the right click menu?

You can use the shortcut <ctrl> <forward-slash> (i.e. <ctrl>/) both pressed together.  This is a standard shortkey on most IDEs so worth learning.  This works on multiple lines too.

OslafEman, 2020-02-18:
Thanks a lot @owennewo, I didn't know that.  As I did not understand your shortcut  (i.e. /), I searched for the usual shortcuts and found the Ctrl K - Ctrl C for comments and Ctrl K - Ctrl U for uncomment.  They work great.

spoenemann, 2020-03-04:
You can press `F1` and then enter parts of a command name (e.g. _comment_) to see the keyboard shortcuts for your operating system / keyboard layout. For example, the _"Toggle Line Comment"_ command is very useful, but its shortcut is quite different on every keyboard layout.

closed 2020-03-04 by spoenemann