arduino/arduino-pro-ide #208: Search on Boards screen is inconsistent

toogooda, 2020-02-10:
partial searches work for some items but for most boards just get excluded no matter what you type.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to boards screen
2. Type 3D
3. The first item "3D printer boards is excluded from the search results

**Expected behavior**
Partial search on any part of board name

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Windows10
 - Version 0.4

**Additional context**
Note I have 50+ boards with ESP, STM and Atmel boards etc so not sure if I have reaching a list limit for the search

kittaakos, 2020-02-10:
Thanks for reporting the issue, @toogooda.

> Go to boards screen

What is the _boards screen_?

Are you using non-official 3rd party boards? If so, perhaps it is due to a missing feature, see here: https://github.com/arduino/arduino-pro-ide/issues/177

toogooda, 2020-02-10:
> What is the _boards screen_?

Think its just called Select Board
 
> Are you using non-official 3rd party boards? If so, perhaps it is due to a missing feature, see here: #177

Yes I have custom boards, If i type ESP32 it shows some but not all for example ESP Wrover disappears from filtered list.

It does seem to be the same as #177

kittaakos, 2020-02-11:
> Select Board

👍

> If i type ESP32 it shows some but not all for example ESP Wrover disappears from filtered list.

Yes, I believe your issue is the same as #177. Thanks for your feedback.

per1234, 2020-02-29:
Closing as duplicate of https://github.com/arduino/arduino-pro-ide/issues/177

closed 2020-02-29 by per1234