arduino/arduino-pro-ide #209: Port Detection does not work

glydeck, 2020-02-11:
I tried this snapshot:
downloads.arduino.cc/arduino-pro-ide/snapshots/arduino-pro-ide-0.0.1-alpha.preview-snapshot-20191021.1-osx.dmg
And the port detection. did not work.
I am using a Mac Book Pro with the 2.3 GHz 8-Core Intel Core i9 processor
System Software is Catalina Version 10.15.3

Hope this helps -- glydeck

per1234, 2020-02-11:
You're using a very outdated version of the Arduino Pro IDE. Please update to the 0.0.4 release, available here:
https://github.com/arduino/arduino-pro-ide/releases
Then let us know whether the problem still occurs.

awgrover, 2020-04-28:
related (old?) #120

awgrover, 2020-04-28:
I have same problem.

arduino-pro-ide 0.0.5
on Ubuntu 18.04.4
I also have Arduino IDE 1.8.10 installed.
Launched `./arduino-pro-ide` from command line, as non-root, the linux-user has `dialout` unix-group.

1.  Started pro ide without any arduino plugged in.
Plugged in Genuine Uno.
No messages in log area.
Went to "select other board and port".
Selected "UNO".
"Ports" pane only shows "NO PORTS DISCOVERED".
"Show all ports" checkbox has no effect.

2. Repeated with UNO plugged in first.
Same: no ports discovered.
linux-terminal output:
```
root WARN please install @theia/electron@0.16.0-next.46a2d510 as a runtime dependency
root INFO Theia app listening on http://localhost:46393.
daemon INFO >>> Starting arduino-cli version: 0.9.0 commit: e4d02b4 daemon from /home/awgrover/dev/thirdparty/arduino-pro-ide-0.0.5-alpha.preview-linux64/resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
daemon INFO Config file not found, using default values
daemon INFO arduino-cli version 0.9.0
daemon INFO Starting daemon on TCP port 50051
daemon INFO Daemon is listening on TCP port 50051...
daemon INFO <<< The 'arduino-cli' daemon is up and running.
root INFO Git askpass helper is listening on http://127.0.0.1:44181.
root INFO Using Git [2.17.1] from the PATH. (/usr/bin/git)
root WARN Failed to resolve module: @types/body-parser
root WARN Failed to resolve module: @types/express
root WARN Failed to resolve module: @types/fs-extra
root WARN Failed to resolve module: @types/lodash.debounce
root WARN Failed to resolve module: @types/lodash.throttle
root WARN Failed to resolve module: @types/react
root WARN Failed to resolve module: @types/react-dom
root WARN Failed to resolve module: @types/react-virtualized
root WARN Failed to resolve module: @types/route-parser
root WARN Failed to resolve module: @types/ws
root WARN Failed to resolve module: @types/yargs
root WARN Failed to resolve module: @theia/editor
root WARN Failed to resolve module: @types/rimraf
root WARN Failed to resolve module: @types/tar-fs
root WARN Failed to resolve module: @types/touch
root WARN Failed to resolve module: @types/uuid
root WARN Failed to resolve module: @typefox/monaco-editor-core
root WARN Failed to resolve module: @theia/monaco
root WARN Failed to resolve module: @theia/preferences
root WARN Failed to resolve module: @theia/navigator
root WARN Failed to resolve module: @types/p-debounce
root WARN Failed to resolve module: @theia/electron
root WARN Failed to resolve module: @theia/messages
root WARN Failed to resolve module: @theia/textmate-grammars
root WARN Failed to resolve module: @types/diff
root WARN Failed to resolve module: @types/p-queue
root WARN Failed to resolve module: @types/dateformat
root WARN Failed to resolve module: @types/google-protobuf
root WARN Failed to resolve module: @types/ps-tree
root WARN Failed to resolve module: @types/react-select
root WARN Failed to resolve module: @types/which
root WARN Collided keybinding is ignored;  {"command":"passthrough","keybinding":"ctrl+k","context":"terminalActive"}  collided with  {"command":"terminal:clear","keybinding":"ctrlcmd+k","context":"terminalActive"}
root WARN Could not register keybinding:
  {"command":"passthrough","keybinding":"ctrl+k","context":"terminalActive"}
Error: "ctrl+k" is in collision with something else [scope:0]
root INFO [nsfw-watcher: 31244] Started watching: /home/awgrover/.theia
root INFO [nsfw-watcher: 31244] Started watching: /home/awgrover/Arduino/every_dev
root INFO  >>> Creating and caching a new client for file:///home/awgrover/Arduino/every_dev...
root INFO [nsfw-watcher: 31244] Started watching: /home/awgrover/Arduino/every_dev
daemon INFO Checking if CLI is Bundled into the IDE
daemon INFO Loading hardware from: /home/awgrover/.arduino15/packages
[... 66 lines cut ...]
daemon INFO Loading hardware from: /home/awgrover/.arduino15/packages
daemon INFO Loading package adafruit from: /home/awgrover/.arduino15/packages/adafruit/hardware
daemon INFO Loaded platform [platform: {ID:"adafruit:avr",Installed:"1.4.13",Latest:"1.4.13"}]
daemon INFO Loading package arduino from: /home/awgrover/.arduino15/packages/arduino/hardware
daemon INFO Loaded platform [platform: {ID:"arduino:avr",Installed:"1.8.2",Latest:"1.8.2",Name:"Arduino AVR Boards"}]
daemon INFO Checking existence of 'tools' path: /home/awgrover/.arduino15/packages/arduino/tools
daemon INFO Loading tools from dir: /home/awgrover/.arduino15/packages/arduino/tools
daemon INFO Loaded tool [tool: Version: 1.3.0 Systems: [OS: i686-linux-gnu, OS: x86_64-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: i686-mingw32]]
daemon INFO Loaded tool [tool: Version: 7.3.0-atmel3.6.1-arduino5 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: x86_64-apple-darwin14, OS: i686-mingw32, OS: i686-linux-gnu, OS: x86_64-linux-gnu]]
daemon INFO Loaded tool [tool: Version: 6.3.0-arduino17 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: x86_64-apple-darwin12, OS: x86_64-linux-gnu, OS: i686-linux-gnu, OS: i686-mingw32]]
daemon INFO Loading package builtin from: /home/awgrover/.arduino15/packages/builtin
daemon INFO Checking existence of 'tools' path: /home/awgrover/.arduino15/packages/builtin/tools
daemon INFO Loading tools from dir: /home/awgrover/.arduino15/packages/builtin/tools
daemon INFO Loaded tool [tool: Version: 5.8-arduino11]
daemon INFO Loaded tool [tool: Version: 1.0.0]
[... cut ...]
daemon INFO Adding libraries dir [dir: /home/awgrover/.arduino15/packages/arduino/hardware/avr/1.8.2/libraries, location: platform]
root INFO  <<< New client has been successfully created and cached for file:///home/awgrover/Arduino/every_dev.
daemon INFO current best: adafruit:avr@1.4.13 [bundle: false, managed: true, version: 1.4.13]
daemon INFO current best: arduino:avr@1.8.2 [bundle: false, managed: true, version: 1.8.2]
Gtk-Message: 11:40:34.942: GtkDialog mapped without a transient parent. This is discouraged.
```

per1234, 2021-02-18:
Hi @awgrover. Thanks so much for the additional information you shared!

Would you mind downloading [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4) and letting us know whether the bug still occurs?

There have been some nice advancements since we discussed this last, so I think the issue might have been fixed along the way.

ubidefeo, 2021-02-25:
@per1234 we haven't had a reply from @glydeck 
Last feedback was 10 months ago.
We believe this is not an issue anymore so let's close it and reopen if it is still an issue

closed 2021-02-25 by ubidefeo