arduino/arduino-pro-ide #210: Language Service crashes on startup (error code 3221225781) - Windows 10 missing dependencies?

owennewo, 2020-02-15:
**Describe the bug**
Newly installed arduino pro 0.0.4 language server crashes (see screenshot and console output below).

Googling error code 3221225781 suggests missing dependencies.  This is a relative clean windows 10 installation and I have no Visual Studio redistributables, are these required? Which version?

**To Reproduce**
Steps to reproduce the behavior:
1. Start Arduino Pro 0.0.4  
2. An empty starter sketch with setup() / loop() is automatically created
3. The following error message is shown in a popup (bottom left)
```Failed to start Arduino language server: Error: Process exited with exit code: 3221225781```

**Expected behavior**
Language service should start, I'm expecting intellisense / completion to work but ctrl space shows "No suggestions".

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 (Build 19041 slow ring)
 - Version 0.0.4 alpha

**Additional context**
```daemon INFO Updating libraries index
root ERROR Unexpected exit code '3221225781' when executing "C:\arduino\resources\app\node_modules\arduino-ide-extension\build\clangd.exe" with args: ["--versn"].
languages ERROR Error occurred while starting language contribution. /services/languages/ino/:id. Error: Process exited with exit code: 3221225781
    at ChildProcess.<anonymous> (C:\arduino\resources\app\node_modules\arduino-ide-extension\lib\node\exec-util.js:124:24)
    at ChildProcess.emit (events.js:182:13)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:240:12)```

owennewo, 2020-02-15:
More info, from git bash I ran the following:
```
$ cd /c/arduino/resources/app/node_modules/arduino-ide-extension/build/
$ ./clangd.exe --version
C:/arduino/resources/app/node_modules/arduino-ide-extension/build/clangd.exe: 
error while loading shared libraries: api-ms-win-crt-locale-l1-1-0.dll: 
cannot open shared object file: No such file or directory```

owennewo, 2020-02-15:
OK - I've fixed this issue by finding the missing DLL
I've installed Visual Studio 2015 redistributable from the microsoft link below and language service within Arduino IDE Pro is now working
https://www.microsoft.com/en-ca/download/confirmation.aspx?id=48145

[image]

Perhaps docs and/or install for windows needs to be updated?

spoenemann, 2020-03-05:
I added the missing DLL. Please reopen if it occurs again.

closed 2020-03-05 by spoenemann