arduino/arduino-pro-ide #212: Randomly a tab kidnaps focus when not selected

ubidefeo, 2020-02-22:
**Describe the bug**
I had a project with several files. Everything worked fine until I toggled Advanced Mode off.
At this point the focus was captured by my last open tab.
The problem presents as impossibility to type or modify text in the currently open tab, which is solved by switching to Advanced Mode.
After a while I switched to all the other tabs one by one and ended up on the last one which at that point contained all the text I had typed while trying to regain access to what I thought was a faulty keyboard.

**To Reproduce**
Steps to reproduce the behavior:
Have not been able to reproduce it yet, but the problem presented itself even after quit-relaunch before I found the issue

**Expected behavior**
n/a

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - OS: MacOS
 - 10.14.6 (Mojave)

**Additional context**
n/a

ubidefeo, 2020-02-22:
I have been messing with it a bit further and it's worse than I thought.
Again, it only happens in basic mode.
See capture video below
https://drive.google.com/file/d/1Z1pU2X30JRqRSq0oape6wi_FBiks8y3T/view?usp=sharing

ubidefeo, 2020-02-22:
Of course when going back to the tabs where text has been edited, no UNDO is available and the file appears to have been automatically saved.
This means your code is lost forever.

spoenemann, 2020-03-05:
Could this be related to https://github.com/eclipse-theia/theia/issues/6298 ?

closed 2020-03-17 by kittaakos