arduino/arduino-pro-ide #213: Failed to start language server. Bundled clangd: File too short

semperErro, 2020-02-26:
Right after starting the IDE, a message occurs in the bottom right saying:

`Failed to start Arduino language server: Error: Process exited with exit code: 127`
 
Terminal output regarding this error:
```
root ERROR Unexpected exit code '127' when executing "/opt/arduino-pro-ide-0.0.4-alpha.preview-linux64/resources/app/node_modules/arduino-ide-extension/build/bin/clangd" with args: ["--version"].
languages ERROR Error occurred while starting language contribution. /services/languages/ino/:id. Error: Process exited with exit code: 127
    at ChildProcess.<anonymous> (/opt/arduino-pro-ide-0.0.4-alpha.preview-linux64/resources/app/node_modules/arduino-ide-extension/lib/node/exec-util.js:124:24)
    at ChildProcess.emit (events.js:182:13)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:240:12)
```

When I try to start the bundled clangd

`/opt/arduino-pro-ide-0.0.4-alpha.preview-linux64/resources/app/node_modules/arduino-ide-extension/build/bin/clangd --version`

it says:

`clangd: error while loading shared libraries: /opt/arduino-pro-ide-0.0.4-alpha.preview-linux64/resources/app/node_modules/arduino-ide-extension/build/bin/../lib/libtinfo.so.5: file too short`

**Desktop (please complete the following information):**
 - OS: Kubuntu
 - Version 19.10

**Additional context**
Running the 0.0.4-alpha.preview

semperErro, 2020-02-29:
Same issue with the new 0.0.5-alpha.preview:

`root ERROR Error executing "/opt/arduino-pro-ide-0.0.5-alpha.preview-linux64/resources/app/node_modules/arduino-ide-extension/build/bin/clangd" --version: /opt/arduino-pro-ide-0.0.5-alpha.preview-linux64/resources/app/node_modules/arduino-ide-extension/build/bin/clangd: error while loading shared libraries: /opt/arduino-pro-ide-0.0.5-alpha.preview-linux64/resources/app/node_modules/arduino-ide-extension/build/bin/../lib/libtinfo.so.5: file too short
languages ERROR Error occurred while starting language contribution. /services/languages/ino/:id. Error: Process failed with error: /opt/arduino-pro-ide-0.0.5-alpha.preview-linux64/resources/app/node_modules/arduino-ide-extension/build/bin/clangd: error while loading shared libraries: /opt/arduino-pro-ide-0.0.5-alpha.preview-linux64/resources/app/node_modules/arduino-ide-extension/build/bin/../lib/libtinfo.so.5: file too short
    at ChildProcess.<anonymous> (/opt/arduino-pro-ide-0.0.5-alpha.preview-linux64/resources/app/node_modules/arduino-ide-extension/lib/node/exec-util.js:126:24)
    at ChildProcess.emit (events.js:182:13)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:240:12)`

spoenemann, 2020-03-04:
It looks like the symbolic links have not been created correctly while unpacking the application's zip file. Could you try opening a terminal in the unpacked application directory and then executing the following?
```
ls -l resources/app/node_modules/arduino-ide-extension/build/lib/
```

If installed correctly, it should display something like this:
```
lrwxrwxrwx 1 user group      14 Feb 26 19:56 libc++abi.so -> libc++abi.so.1
lrwxrwxrwx 1 user group      16 Feb 26 19:56 libc++abi.so.1 -> libc++abi.so.1.0
-rw-r--r-- 1 user group  308288 Feb 26 18:01 libc++abi.so.1.0
-rw-r--r-- 1 user group      28 Feb 26 18:06 libc++.so
lrwxrwxrwx 1 user group      13 Feb 26 19:56 libc++.so.1 -> libc++.so.1.0
-rw-r--r-- 1 user group 1017080 Feb 26 18:01 libc++.so.1.0
lrwxrwxrwx 1 user group      15 Feb 26 19:56 libtinfo.so.5 -> libtinfo.so.5.9
-rwxr-xr-x 1 user group  193928 Feb 26 18:01 libtinfo.so.5.9
```

The `->` indicate symbolic links. If these arrows are missing, something went wrong when unpacking the archive.

semperErro, 2020-03-05:
`ls -l resources/app/node_modules/arduino-ide-extension/build/lib/`

outputs:

```
-rwxrwxrwx 1 foo foo      14 Feb 26 18:06 libc++abi.so
-rwxrwxrwx 1 foo foo      16 Feb 26 18:06 libc++abi.so.1
-rw-r--r-- 1 foo foo  308288 Feb 26 18:01 libc++abi.so.1.0
-rw-r--r-- 1 foo foo      28 Feb 26 18:06 libc++.so
-rwxrwxrwx 1 foo foo      13 Feb 26 18:06 libc++.so.1
-rw-r--r-- 1 foo foo 1017080 Feb 26 18:01 libc++.so.1.0
-rwxrwxrwx 1 foo foo      15 Feb 26 18:06 libtinfo.so.5
-rwxr-xr-x 1 foo foo  193928 Feb 26 18:01 libtinfo.so.5.9
```

So there must have went something wrong when unpacking the archive using KUbuntu Ark.

I tried unpacking the archive with `unzip` instead of Ark and then the symbolic links where correct:

```
lrwxrwxrwx 1 foo foo      14 Mär  5 12:12 libc++abi.so -> libc++abi.so.1
lrwxrwxrwx 1 foo foo      16 Mär  5 12:12 libc++abi.so.1 -> libc++abi.so.1.0
-rw-r--r-- 1 foo foo  308288 Feb 26 18:01 libc++abi.so.1.0
-rw-r--r-- 1 foo foo      28 Feb 26 18:06 libc++.so
lrwxrwxrwx 1 foo foo      13 Mär  5 12:12 libc++.so.1 -> libc++.so.1.0
-rw-r--r-- 1 foo foo 1017080 Feb 26 18:01 libc++.so.1.0
lrwxrwxrwx 1 foo foo      15 Mär  5 12:12 libtinfo.so.5 -> libtinfo.so.5.9
-rwxr-xr-x 1 foo foo  193928 Feb 26 18:01 libtinfo.so.5.9

```
The clangd error does not appear anymore.

closed 2020-03-05 by semperErro