arduino/arduino-pro-ide #214: Problem with WiFiNINA.h

GitHubRea, 2020-02-26:
**Describe the bug**
problem with WiFi-Status constant.
see screenshot: arduino-pro-ide 0.0.5
[image]

But in Arduion 1.8.12 it is working fine
[image]

**Desktop (please complete the following information):**
 - OS: [macOS]
 - Version [10.15.3]

ubidefeo, 2020-02-26:
hi @GitHubRea 
Not really clear what the issue is.
If you are referring to the constant not being recognised this might be a minor issue.
Does the code compile and run properly?
If it does not compile and run then it's indeed a bug :)

Also, could you post the code in a snippet?
that allows us to test it doing copy/paste :)

GitHubRea, 2020-02-27:
Hello ubidefeo
sorry the delay.

here the code.
Thx

`
#include <SPI.h>
#include <WiFiNINA.h>

char ssid[] = "xxx";
char pass[] = "xxx";
int status = WL_IDLE_STATUS;
byte mac[] = { 0x90, 0xA2, 0xDA, 0x0D, 0x7D, 0x54 };

WiFiServer server(80);

void setup() {
  // initialize serial:
  Serial.begin(9600);
  Serial.println("Attempting to connect to WPA network...");
  Serial.print("SSID: ");
  Serial.println(ssid);

  status = WiFi.begin(ssid, pass);
  if ( status != WL_CONNECTED) { 
    Serial.println("Couldn't get a wifi connection");
    while(true);
  } 
  else {
    server.begin();
    Serial.print("Connected to wifi. My address:");
    IPAddress myAddress = WiFi.localIP();
    Serial.println(myAddress);

  }
}

void loop(void) {
WiFiClient client = server.available();

//two variables to hold your temperatures
float temperatureIndoor;
float temperatureOutdoor;

if (client) {
    Serial.println("new client");
    //an http request ends with a blank line
    boolean currentLineIsBlank = true;
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        Serial.write(c);
        // if you've gotten to the end of the line (received a newline
        // character) and the line is blank, the http request has ended,
        // so you can send a reply
        if (c == '\n' && currentLineIsBlank) {
          // send a standard http response header
          client.println("HTTP/1.1 200 OK");
          client.println("Content-Type: text/html");
          client.println("Connnection: close");
          client.println();
          
          temperatureIndoor = 22.77;
          temperatureOutdoor = 15.55;
        
          client.print("{\"arduino\":[{\"location\":\"indoor\",\"celsius\":\"");
          client.print(temperatureIndoor);
          client.print("\"},");
          client.print("{\"location\":\"outdoor\",\"celsius\":\"");
          client.print(temperatureOutdoor);
        
          client.print("\"}]}");
          client.println();
          break;
        }
        if (c == '\n') {
          currentLineIsBlank = true;
        } 
        else if (c != '\r') {
          //you've gotten a character on the current line
          currentLineIsBlank = false;
        }
      }
    }
    // give the web browser time to receive the data
    delay(1);
    // close the connection:
    client.stop();
    Serial.println("client disonnected");
  }

}
`

ubidefeo, 2020-02-27:
@GitHubRea 
It is working for me out of the box.
If you cannot compile or the IDE cannot see library definitions it might mean that you haven't installed any cores yet.
Check the README on the main GitHub page for this project and you'll see.
Let me know if this fixes your issue

>The Arduino Pro IDE does not provide pre-installed cores, to have board support you have to install a core by using the menu

> Tools -> Board Manager...

GitHubRea, 2020-02-28:
Hello ubidefeo,
you are right, it is not possible to compile. But I installed the needed core and library.(see attachments)
I am using the MKR Wifi 1010 port. 

[ArduinoProIde_1.pdf](https://github.com/arduino/arduino-pro-ide/files/4268811/ArduinoProIde_1.pdf)
[ArduinoProIde_2.pdf](https://github.com/arduino/arduino-pro-ide/files/4268813/ArduinoProIde_2.pdf)

Thanks for the effort!

ubidefeo, 2020-03-01:
unfortunately I cannot reproduce the issue, 
let's see if someone else has an idea of what could be troubling your setup

GitHubRea, 2020-03-01:
Hello I am so sorry,

I found the problem for my issue.
I saved the project to an other folder, not in the Arduino basic root.
[ArduinoIdePro_library.pdf](https://github.com/arduino/arduino-pro-ide/files/4271815/ArduinoIdePro_library.pdf)
And in this project folder there is no library folder included.
I copied the library folder in this project folder. After that it was compiling.

So I think I should handle my projects in the installed root of Arduino.
[ArduinoIdePro_correctFolder.pdf](https://github.com/arduino/arduino-pro-ide/files/4271836/ArduinoIdePro_correctFolder.pdf)
What do you think?

Thanks a lot!

GitHubRea, 2020-03-01:
... is it possible to set the folder root of library somewhere in the ProIDE.
Because in standard IDE it is possible to save setches in other folders.
Thx and sorry for my boor english.

ubidefeo, 2021-02-19:
@GitHubRea 
I know it's been a long time, but this works for me on the latest Alpha 0.1.4

we're cleaning up old issues.
@per1234 I saw you can still replicate this, I have noticed that as soon as you trigger the Language Server by selecting a board it starts working for me