arduino/arduino-pro-ide #215: alpha5 does not see preinstalled boards

joysfera, 2020-02-26:
**Describe the bug**
alpha5 cannot see any of the preinstalled boards that are visible in alpha4

**To Reproduce**
1. run alpha5
2. check installed boards

**Expected behavior**
All preinstalled boards are visible and usable.

**Desktop (please complete the following information):**
 - Ubuntu
 - 18.04

joysfera, 2020-02-26:
It's actually messed even more, somehow. The standard Arduino boards are preinstalled but don't work. Installing AVR board support (v1.8.2) worked but then again program couldn't be compiled for AVR. So even reinstalling boards doesn't seem to help.

Going back to 0.0.4alpha fixes everything - installed boards re-appear and software can be compiled.

Master811129, 2020-02-27:
i have same problem the IDE doesn't show any custom board.😠

Master811129, 2020-02-27:
i can see my board in "arduino: daemon" when i search some boards.

> [INFO] current best: MightyCore:avr@2.0.3 [bundle: false, managed: true, version: 2.0.3]
[INFO] current best: esp32:esp32@1.0.4 [bundle: false, managed: true, version: 1.0.4]
[INFO] current best: esp32:esp32@1.0.4 [bundle: false, managed: true, version: 1.0.4]
[INFO] current best: ATTinyCore:avr@1.3.3 [bundle: false, managed: true, version: 1.3.3]
[INFO] current best: MightyCore:avr@2.0.3 [bundle: false, managed: true, version: 2.0.3]
[INFO] current best: arduino:nrf52@1.0.2 [bundle: false, managed: true, version: 1.0.2]
[INFO] current best: arduino:avr@1.8.2 [bundle: false, managed: true, version: 1.8.2]
[INFO] current best: arduino:megaavr@1.8.5 [bundle: false, managed: true, version: 1.8.5]
[INFO] current best: MiniCore:avr@2.0.3 [bundle: false, managed: true, version: 2.0.3]
[INFO] current best: diy_attiny:avr@2018.9.9 [bundle: false, managed: true, version: 2018.9.9]

Master811129, 2020-02-27:
OK. It seems i cannot see them in boards manager.

Master811129, 2020-02-27:
yup! It's a cli bug. I can fix this issue by downgrading arduino-cli in path of the IDE.

Tuzi07, 2020-02-29:
Hey Master811129, can you share how you solved it?

Master811129, 2020-03-01:
> Hey Master811129, can you share how you solved it?
> Bug persists after going back do 0.0.4
> [INFO] current best: esp32:esp32@1.0.4 [bundle: false, managed: true, version: 1.0.4]

I replaced an older version of arduino-cli (0.7.0) to this path:
[Installed folder]\resources\app\node_modules\arduino-ide-extension\build
@kittaakos @spoenemann  @ubidefeo do you have any idea?

kittaakos, 2020-03-02:
Thanks for naming me, @Master811129. I am checking what is going on...

kittaakos, 2020-03-02:
Please help me out a bit. I did the followings:
 - Downloaded the `0.0.5` Pro IDE and wiped the user and data folders (`rm -rf ~/Library/Arduino15/ && rm -rf ~/Documents/Arduino/`).
 - I started the app, copy-pasted the _Blink_ example, attached my Uno (replica) board; I could verify and upload.
 - Then I created a new sketch, copy-pasted the _Blink_ example again, installed the _Arduino SAMD Boards (32-bits ARM Cortex-M0+)_ core package, attached my MKR1000 board; I could verify and upload.

I think there two separate issues here:

> The standard Arduino boards are preinstalled but don't work. Installing AVR board support (v1.8.2) worked but then again program couldn't be compiled for AVR.

@joysfera, can you mean by does not work?

> i have same problem the IDE doesn't show any custom board.😠

@Master811129, #177 was not part of `0.0.5`.

Tuzi07, 2020-03-02:
Replaced arduino-cli to 0.7.0, bug persists

joysfera, 2020-03-02:
@kittaakos please ignore my comment about issues with compiling for AVR (Arduino Uno). I cannot reproduce it right now. I admit I have a newer AVR core installed than I had originally so that might have an effect. Anyway, just forget that, seems like the AVR target is OK out of the box.

kittaakos, 2020-03-02:
> seems like the AVR target is OK out of the box.

Thank you, @joysfera

@Tuzi07, can you please give us some details about what did you try, what is wrong and what would you expect? Thanks!

joysfera, 2020-03-02:
@kittaakos and @Tuzi07 , you wanted to see how to fix the 0.0.5alpha by downgrading arduino-cli? Look here:
```
joy@amd:/tmp/005$ find . -name arduino-cli
./resources/app/node_modules/arduino-ide-extension/build/arduino-cli
joy@amd:/tmp/005$ cd ./resources/app/node_modules/arduino-ide-extension/build/
joy@amd:/tmp/005/resources/app/node_modules/arduino-ide-extension/build$ rm arduino-cli 
joy@amd:/tmp/005/resources/app/node_modules/arduino-ide-extension/build$ ln -s /home/joy/bin/arduino-cli 
```

This fixes the issue. Unfortunately I don't remember which version of `arduino-cli` I've got installed and it does not print its own version so all I can do to help you identify it is to show you its file length:

```
joy@amd:~$ l bin/arduino-cli 
-rwxr-xr-x 1 joy joy 16015360 pro 31 12:23 bin/arduino-cli*
```

kittaakos, 2020-03-02:
> @kittaakos and
> , you wanted to see how to fix the 0.0.5alpha by downgrading arduino-cli?

No, I wanted to understand what the problem is 😊

Tuzi07, 2020-03-02:
1) Installed Arduino IDE beta, `arduino-1.9.0-beta-linux64`
2) Went to preferences and added `https://dl.espressif.com/dl/package_esp32_index.json` in additional boards managers URLs to enable `esp32` in Boards Manager
3) Installed `esp32` in Boards Manager
4) Everything working in Arduino IDE as expected
5) Installed `arduino-cli`
`brew update
brew install arduino-cli`
6) `arduino-cli` file is in `home/.local/bin` (Ubuntu 18.04)
7) Downloaded `arduino-pro-ide-0.0.5-alpha.preview-linux64`, extracted and ran `arduino-pro-ide` in terminal
8) Arduino Pro IDE opens and there is the following output:
`[INFO] current best: esp32:esp32@1.0.4 [bundle: false, managed: true, version: 1.0.4]`
9) Clicked in ´no board selected´ and typed ´esp32´ in ´SEARCH BOARD´, nothing shows up.

Tuzi07, 2020-03-02:
@joysfera I tried doing what you said, but when I open Arduino Pro I get <no output yet>, maybe arduino-cli was not installed properly, how do I confirm this?

joysfera, 2020-03-02:
@Tuzi07 follow the instructions for `arduino-cli` to install it properly first. You want the ```arduino-cli board listall``` to list all installed cores - only then it makes sense to continue with `arduino-pro-ide`.

Tuzi07, 2020-03-03:
Result for `arduino-cli board listall`:
```
Board Name                      FQBN                                                                                          
AI Thinker ESP32-CAM            esp32:esp32:esp32cam                       
ALKS ESP32                      esp32:esp32:alksesp32                      
Adafruit ESP32 Feather          esp32:esp32:featheresp32                   
BPI-BIT                         esp32:esp32:bpi-bit                        
D-duino-32                      esp32:esp32:d-duino-32                     
DOIT ESP32 DEVKIT V1            esp32:esp32:esp32doit-devkit-v1            
Dongsen Tech Pocket 32          esp32:esp32:pocket_32                      
ESP32 Dev Module                esp32:esp32:esp32                          
ESP32 FM DevKit                 esp32:esp32:fm-devkit                      
ESP32 Pico Kit                  esp32:esp32:pico32                         
ESP32 Wrover Module             esp32:esp32:esp32wrover                    
ESP32vn IoT Uno                 esp32:esp32:esp32vn-iot-uno                
ESPea32                         esp32:esp32:espea32                        
ESPectro32                      esp32:esp32:espectro32                     
Electronic SweetPeas - ESP320   esp32:esp32:esp320                         
FireBeetle-ESP32                esp32:esp32:firebeetle32                   
Frog Board ESP32                esp32:esp32:frogboard                      
Heltec WiFi Kit 32              esp32:esp32:heltec_wifi_kit_32             
Heltec WiFi LoRa 32             esp32:esp32:heltec_wifi_lora_32            
Heltec WiFi LoRa 32(V2)         esp32:esp32:heltec_wifi_lora_32_V2         
Heltec Wireless Stick           esp32:esp32:heltec_wireless_stick          
Hornbill ESP32 Dev              esp32:esp32:hornbill32dev                  
Hornbill ESP32 Minima           esp32:esp32:hornbill32minima               
IntoRobot Fig                   esp32:esp32:intorobot-fig                  
LOLIN D32                       esp32:esp32:d32                            
LOLIN D32 PRO                   esp32:esp32:d32_pro                        
LoPy                            esp32:esp32:lopy                           
LoPy4                           esp32:esp32:lopy4                          
M5Stack-Core-ESP32              esp32:esp32:m5stack-core-esp32             
M5Stack-FIRE                    esp32:esp32:m5stack-fire                   
M5Stick-C                       esp32:esp32:m5stick-c                      
MH ET LIVE ESP32DevKIT          esp32:esp32:mhetesp32devkit                
MH ET LIVE ESP32MiniKit         esp32:esp32:mhetesp32minikit               
MagicBit                        esp32:esp32:magicbit                       
Microduino-CoreESP32            esp32:esp32:CoreESP32                      
Nano32                          esp32:esp32:nano32                         
Node32s                         esp32:esp32:node32s                        
NodeMCU-32S                     esp32:esp32:nodemcu-32s                    
Noduino Quantum                 esp32:esp32:quantum                        
ODROID ESP32                    esp32:esp32:odroid_esp32                   
OLIMEX ESP32-DevKit-LiPo        esp32:esp32:esp32-DevKitLipo               
OLIMEX ESP32-EVB                esp32:esp32:esp32-evb                      
OLIMEX ESP32-GATEWAY            esp32:esp32:esp32-gateway                  
OLIMEX ESP32-PoE                esp32:esp32:esp32-poe                      
OLIMEX ESP32-PoE-ISO            esp32:esp32:esp32-poe-iso                  
OROCA EduBot                    esp32:esp32:oroca_edubot                   
Onehorse ESP32 Dev Module       esp32:esp32:onehorse32dev                  
Pycom GPy                       esp32:esp32:gpy                            
Silicognition wESP32            esp32:esp32:wesp32                         
SparkFun ESP32 Thing            esp32:esp32:esp32thing                     
SparkFun LoRa Gateway 1-Channel esp32:esp32:sparkfun_lora_gateway_1-channel
T-Beam                          esp32:esp32:t-beam                         
TTGO LoRa32-OLED V1             esp32:esp32:ttgo-lora32-v1                 
TTGO T-Watch                    esp32:esp32:ttgo-t-watch                   
TTGO T1                         esp32:esp32:ttgo-t1                        
ThaiEasyElec's ESPino32         esp32:esp32:espino32                       
TinyPICO                        esp32:esp32:tinypico                       
Turta IoT Node                  esp32:esp32:turta_iot_node                 
VintLabs ESP32 Devkit           esp32:esp32:vintlabs-devkit-v1             
WEMOS D1 MINI ESP32             esp32:esp32:d1_mini32                      
WEMOS LOLIN32                   esp32:esp32:lolin32                        
WeMos WiFi&Bluetooth Battery    esp32:esp32:WeMosBat                       
WiPy 3.0                        esp32:esp32:wipy3                          
Widora AIR                      esp32:esp32:widora-air                     
XinaBox CW02                    esp32:esp32:cw02               
            
u-blox NINA-W10 series (ESP32)  esp32:esp32:nina_w10
```
Isn't it installed properly?

kittaakos, 2020-03-03:
> 1. Installed Arduino IDE beta, `arduino-1.9.0-beta-linux64`

You do not need this. The Arduino IDE and the Pro IDE are different things.

> 2\. Went to preferences and added `https://dl.espressif.com/dl/package_esp32_index.json` in additional boards managers URLs to enable `esp32` in Boards Manager

This is work in progress (#177)

> 5\. Installed `arduino-cli`
>     `brew update`
>     `brew install arduino-cli`

You do not have to. The Pro IDE contains the CLI. 

> I tried doing what you said, but when I open Arduino Pro I get `<no output yet>`, maybe arduino-cli was not installed properly, how do I confirm this?

You can confirm it by starting `./arduino-pro-ide` from the terminal. Do you see errors? Do you see the `arduino-cli` running as a process? You can also try to reset the layout in the Pro IDE: <kbd>Ctrl/Cmd</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd> or <kbd>F1</kbd> and type `Reset Workbench Layout`.

kittaakos, 2020-03-03:
> but when I open Arduino Pro I get `<no output yet>`, maybe arduino-cli was not installed properly

Do you see an `arduino-cli` process running when the Pro IDE does not run?

Master811129, 2020-03-03:
> Please help me out a bit. I did the followings:
>
> * Downloaded the `0.0.5` Pro IDE and wiped the user and data folders (`rm -rf ~/Library/Arduino15/ && rm -rf ~/Documents/Arduino/`).
> * I started the [...]

umm i think "MKR1000 board" is an official board I have problem with 3RD PARTY (non-official) cores:
To Reproduce:
1.Install some 3RD PARTY core like :ESP32 ,Mighty Core, Mini Core ,Micro Core using JAVA IDE.
2.open Arduino Pro IDE 0.0.5 alpha (it must be 0.0.5 we don't have this problem in 0.0.4).
3.you can't see them in the boards and ports menu (you cannot see them in boards manager.)

kittaakos, 2020-03-03:
> umm i think "MKR1000 board" is an official board I have problem with 3RD PARTY (non-official) cores:

@Master811129, the original title was: _alpha5 does not see preinstalled boards_

> please ignore my comment about issues with compiling for AVR (Arduino Uno). I cannot reproduce it right now. I admit I currently have a newer AVR core installed than I had originally so that might have an effect. Anyway, just forget that, seems like the AVR target is OK out-of-the-box now (in 0.0.5alpha).

We figured out, it is about the 3rd party cores.

You have opened #177, you know exactly, it is not yet supported with the current release (v0.0.5).

I am going to write down a short step by step instruction and add it to this thread, so you can force the v0.0.5 version to make the 3rd party core packages work. Be prepared, this is very much insider, but as I see you are eagerly looking for the 3rd party boards support 😊

joysfera, 2020-03-03:
@kittaakos thanks. 

BTW, until #10 is implemented the Arduino Pro IDE cannot be used most of the time (unless you are lucky to be OK with the default parameters).

kittaakos, 2020-03-03:
> until #10 is implemented

I started to work on that feature today 😊 I do my best to make sure you get all these much-requested features ASAP.

Tuzi07, 2020-03-13:
Tried things out on Windows, 0.0.4 works

Master811129, 2020-05-05:
NOW IT WORKS IN 0.0.6
thank you @kittaakos. 😉

closed 2020-05-05 by kittaakos