arduino/arduino-pro-ide #216: Alpha 0.0.5: Provide an example for configuring the debugger

rei-vilo, 2020-02-28:
**Is your feature request related to a problem? Please describe.**
Alpha 0.0.5 brings debugging to Arduino IDE Pro. But, how to configure it?

**Describe the solution you'd like**
Provide an example for configuring the debugger.

**Describe alternatives you've considered**
Wait for the next release!

**Additional context**
With an Arduino Zero board, programming USB port.

+ Error message
[image]

+ Default configuration
[image]

per1234, 2020-02-28:
@rei-vilo the error:
```
Error during Debug: Cannot get command line for tool: cannot get programmer tool: undefined 'debug.tool' property
```
indicates you are using a version of Arduino SAMD Boards prior to 1.8.5. Support for the debugger feature was only added in the Arduino SAMD Boards 1.8.5 release (https://github.com/arduino/ArduinoCore-samd/commit/e6af98a744637d19bd9896150cbddf36b06e2883).

So use Boards Manager to update Arduino SAMD Boards to 1.8.5 and you should be able to get started debugging with your Arduino Zero!

If you have any problems after that, let me know. I'm able to use the onboard debugger on the Zero with the Arduino Pro IDE, so I'll probably be able to help you out.

rei-vilo, 2020-02-28:
Thank you for your answer. I’ll give a try and let you know.

rei-vilo, 2020-02-28:
It works with Arduino SAMD Boards 1.8.5.

I expected the Arduino Pro IDE to check the boards packages were up to date.

closed 2020-02-28 by rei-vilo

Team4513, 2020-09-29:
May board manager only lists versions up to 1.8.3

per1234, 2020-09-29:
>My board manager only lists versions up to 1.8.3

Are you referring to the "Arduino SAMD Boards (32-bits ARM Cortex-M0+)" platform?

@Team4513 I suspect you're looking at the Arduino AVR Boards platform. If so, it's correct that it only lists versions of that platform up to 1.8.3 because the latest release is 1.8.3.

There is no support in the Arduino AVR Boards platform for the debugging feature, nor are there any plans for adding support. However, if you have one of the Arduino SAMD Boards (e.g., Zero, MKR1000), you can use the debugging feature with it.

ubidefeo, 2020-09-29:
@Team4513 there isn't much to be disappointed about: AVR Mega (not AVR X Mega) does not really support debug unless you get DebugWIre ICE programmers and go through the pain of not having a RESET pin as well as learning how to do debug over the DebugWIre using command line tools which, let's face it, are obscure to most people who have not digged into the meanders of AVR. I remember the awe when I was able to send a command to my UNO to stop execution of my program and start it again :D 
You can read something about DebugWire here, especially the fact that it is neither documented nor open-source.

Some more advanced AVR boards have JTAG, and I'm sure someone has made a platform supporting that using Atmel ICE or JLINK, but I can't mention one out the top of my head, I'm sure googling might bare fruits :)

That said, everyone is allowed to extend our cores to implement debugging capabilities for debug-enabled microcontrollers.
In these posts you can find relevant answers along the same line we gave to other users
* #328 
* #301

Team4513, 2020-09-30:
Thanks, I was wondering about that. Any chance the Due will eventually have
debug support?