arduino/arduino-pro-ide #217: Alpha 5 Debugging: breakpoints on local libraries are not considered

rei-vilo, 2020-02-28:
**Describe the bug**
Breakpoints defined on local libraries are not considered.

**To Reproduce**
Very basic example with 3 files: `Blink.ino`, `LocalLibrary.h`, `LocalLibrary.cpp`
Two breakpoints defined: one in `Blink.ino`, another in `LocalLibrary.cpp`

**Expected behavior**
Breakpoints defined on local library `LocalLibrary.cpp` should be considered.

Actually, all the files are duplicated under a temporary folder, `/private/var/folders/wn/n7qqb8ss0k3bvpqwfcdwp_7r0000gn/T/arduino-sketch-7CD787D3756B222A4A3572C34ED19F79/sketch/`

The breakpoint on `Blink.ino` is kept, but not the breakpoint on `LocalLibrary.cpp`.

**Workaround**

Launch the debugging session.
Define the breakpoint on the local library again.

**Screenshots**

+ Initial project folder during editing

[image]

+ Initial project folder during debugging

[image]

+ Copy to temporary folder during debugging

[image]

**Desktop (please complete the following information):**
 - macOS 10.15
 - Arduino Pro IDE 0.0.5

** Code

+ Blink.ino

```
#include "LocalLibrary.h"
uint8_t myLED;

void setup()
{
    Serial.begin(9600);
    delay(500);
    Serial.println(__FILE__);

// !!! Help: http://bit.ly/2BkhLB6

    myLED = LED_BUILTIN;

    pinMode(myLED, OUTPUT);
}

void loop()
{
    Serial.println(millis());

    blinkLED(myLED, 3, 333);
    delay(1000);
}
```

+ LocalLibrary.h

```
#include "Arduino.h"

#ifndef blink_LocalLibrary_h
#define blink_LocalLibrary_h

///
/// @brief      Blink a LED
/// @details	LED attached to pin is turned on then off
/// @note       Total cycle duration = ms
/// @param      pin pin to which the LED is attached
/// @param      times number of times
/// @param      ms cycle duration in ms
/// @param		level level for HIGH, default=true=positive logic, false=negative logic
///
void blinkLED(uint8_t pin, uint8_t times, uint16_t ms, bool level = true);
// !!! Help: https://bit.ly/2Bwmyk6

#endif // blink_LocalLibrary_h
```

+ LocalLibrary.cpp

```
#include "LocalLibrary.h"

// Code
void blinkLED(uint8_t pin, uint8_t times, uint16_t ms, bool level)
{
    for (uint8_t i = 0; i < times; i++)
    {
        digitalWrite(pin, level ? HIGH : LOW);
        delay(ms >> 1);
        digitalWrite(pin, level ? LOW : HIGH);
        delay(ms >> 1);
    }
}
```

spoenemann, 2020-03-04:
I think this must be fixed in the CLI. I created an issue here:
https://github.com/arduino/arduino-cli/issues/601

rei-vilo, 2020-03-04:
Thank you.

Waiting for Alpha 0.06 now!

rei-vilo, 2020-09-02:
Unfortunately, same issue with Alpha 0.1.1.

[image]

The breakpoint defined on line 27 of `LocalLibrary.cpp` labelled `LocalLibrary.cpp .../Users/...`...

[image]

...isn't used on `LocalLibrary.cpp .../Private/...` when debugging

[image]

rei-vilo, 2020-10-13:
See #366

ubidefeo, 2020-10-13:
@rei-vilo 
For the next minor alpha release we planned on addressing a lot of debugger related issues, and this is one of them.
Cannot give you an ETA, but I can say that it won't be 2 months ;)

rei-vilo, 2020-10-13:
@ubidefeo Thank you, great to know!

I know software development isn't easy, especially when targeting so many boards...

ubidefeo, 2021-02-18:
@rei-vilo
would you care to give a shot at the last released alpha? (0.1.4)
we have brought a lot of enhancements to the debugger

Will close it because I cannot reproduce it anymore, but feel free to reopen if you're still affected.

ubidefeo, 2021-02-18:
Seems like @per1234 can still reproduce it, hence reopening

rei-vilo, 2021-02-18:
Sure, I plan to try the new release with the same project.

rei-vilo, 2021-02-18:
@ubidefeo After searching a bit for the procedure, I finally found it here at https://github.com/arduino/arduino-pro-ide/issues/366#issuecomment-775734448, and I ran the very same example.

Unfortunately, only the breakpoint on the main sketch is considered. The breakpoint on the library is listed, but not active.

[image]

per1234, 2021-02-18:
>@per1234 can still reproduce it

That's correct. When I follow rei-vilo's instructions from https://github.com/arduino/arduino-pro-ide/issues/217#issue-572776322, the breakpoint in LocalLibrary.cpp is grayed out in the debugger's "Breakpoints" section, and is an empty circle rather than the expected red dot in the sketch tab. The breakpoint's tooltip says this:
>No source file named c:\Users\per\Documents\Arduino\debug\NonInoTab\LocalLibrary.cpp. (from break-insert "c:\Users\per\Documents\Arduino\debug\NonInoTab\LocalLibrary.cpp:8")

If I add a breakpoint to the .ino file, it works perfectly.

ubidefeo, 2021-02-18:
@rsora and I also got to that conclusion and opened an internal task which @cmaglie and @kittaakos are going to look into.
Essentially anything without the `.ino` extension is referenced from the temp folder, so it opens in a new tab.
You can add a breakpoint to that file and it will work fine.
The original source file is simply ignored.
It's very likely just a mapping issue, and I think @cmaglie can fix that.
It's a good bug 😬