arduino/arduino-pro-ide #218: Problem with high contrast theme: icons in top-left are invisible.

Master811129, 2020-02-28:
**Describe the bug**
icons in top-left are invisible.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings'
2. Click on 'Color Theme'
4. See error

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 -Arduino version: 0.0.5 alpha

kittaakos, 2020-04-28:
This has been fixed.

[image]

Master811129, 2020-04-28:
> This has been fixed.
> 
> [image]

cool can we see the fix in 0.0.5?

closed 2020-04-28 by Master811129

Master811129, 2020-05-05:
[image]

in 0.0.6 still icons in top-left are invisible.

kittaakos, 2020-05-05:
> in 0.0.6 still icons in top-left are invisible.

I know. It did not make it to the 0.0.6, sorry. The fix is already on our `master`, though.