arduino/arduino-pro-ide #219: Error during Debug: Cannot get command line for tool: cannot get programmer tool: undefined 'debug.tool' property

Matszwe02, 2020-02-28:
HI. I just downloaded Arduino Pro, uploaded a program into my leonardo and hit "debug", and the error appears.

rei-vilo, 2020-02-28:
It seems only boards based on the SAMD MCU with a debugger are supported.

kampfk3ks, 2020-02-29:
is there a list of supported Boards? I tried with  Uno, nano and  Mega and nothing worked so far.

rei-vilo, 2020-02-29:
Changelog:

+ Debugger support for SAMD boards

Matszwe02, 2020-02-29:
Will it support 8 bit Atmega boards?

rei-vilo, 2020-02-29:
You need a debugger. Only the Arduino Zero board includes a debugger for the moment. The Arduino Nano 33 provisions pads for a SWD connection to an external debugger.

ubidefeo, 2020-02-29:
@Matszwe02 
debugging 8bit AVR involves using a protocol called DebugWire.
Right now we support SAMD chips, and since DebugWire is a complex thing to work with, I can't see it being supported.
You also need either an Atmel ICE, STK500 or Dragon

el-samiyel, 2020-03-15:
Even with the atmel ICE the message remains: Error during Debug: Cannot get command line for tool: cannot get programmer tool: undefined 'debug.tool' property

Does anything need to be done in a config?

ubidefeo, 2020-03-15:
@el-samiyel which board are you testing this with?

el-samiyel, 2020-03-15:
I am using the M0. Considering it doesn't have a edbg can't the atmel ice via swd be used in its place? Thanks

ubidefeo, 2020-03-15:
could you please provide a link to the product?
of course if you have SWD pins exposed you can debug, but need the board in question to have the extra config settings defined within boards.txt of its core

rei-vilo, 2020-03-15:
I tried first with the Arduino M0 Pro from Arduino.org and it didn’t work. 

Then, with the Arduino Zero from Arduino.cc / Genuino, debugging worked fine.

el-samiyel, 2020-03-15:
Hi @ubidefeo  Yes, just as @rei-vilo has said. I used an Arduino M0, this is the same as the Arduino zero but without the edgb. So I plug in my atmel ICE into the SWD connector but am presented with the message mentioned above.

Its as if the PRO IDE doesn't see the Atmal ICE? However, what extra parameters should be set in the boards.txt?

Board: [https://store.arduino.cc/arduino-m0](url)

Thanks

ericklein, 2020-05-31:
What boards and external debuggers are currently supported and is there a configuration guide available? I've tried an Adafruit M0 Express via JLINK and get the error "https://adafruit.github.io/arduino-board-index/package_adafruit_index.json" and when I try a MKR1000 via JLINK I get the error "Error: unable to find CMSIS-DAP device".

tomsihap, 2020-09-04:
> You need a debugger. Only the Arduino Zero board includes a debugger for the moment. The Arduino Nano 33 provisions pads for a SWD connection to an external debugger.

Could other boards (Arduino Uno for example) be supported in the future or is it technically not possible because of this lack of built-in debugger ?

ubidefeo, 2020-09-04:
@tomsihap 
the ATMega328 (on the Arduino UNO) only supports debugging via WireDebug interface, which needs to be enabled at low level using fuses and is only supported by a bunch of programmers and has never really been successful.
We will support interfaces based on GDB and OpenOCD for our own boards, as well hybrid debugging over USB like we're experimenting with the Portenta H7.
Debugging on each platform requires implementation by the platform supplier, with a series of debug scripts that launch the required tools, which also have to be installed based on platform specification.

You can read about DebugWire here http://www.ruemohr.org/docs/debugwire.html out of curiosity, but I doubt anyone will ever put effort into implementing tooling for that

tuxedo0801, 2020-09-27:
same issue here as @el-samiyel described ... using M0 board with Atmel ICE SAM debugger. does not work. same error here.

ubidefeo, 2020-09-27:
@tuxedo0801 
that board is lacking the debug line in the platform
when you read the `boards.txt` file in the hardware platform definition package, you can see that while the `MKRZERO` and other boards have this block
```
mkrzero.debug.tool=gdb-openocd
mkrzero.upload.tool=bossac
mkrzero.upload.protocol=sam-ba
mkrzero.upload.maximum_size=262144
mkrzero.upload.maximum_data_size=32768
mkrzero.upload.use_1200bps_touch=true
mkrzero.upload.wait_for_upload_port=true
mkrzero.upload.native_usb=true
mkrzero.build.mcu=cortex-m0plus
mkrzero.build.f_cpu=48000000L
mkrzero.build.usb_product="Arduino MKRZero"
mkrzero.build.usb_manufacturer="Arduino LLC"
mkrzero.build.board=SAMD_MKRZERO
mkrzero.build.core=arduino
mkrzero.build.extra_flags=-DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ {build.usb_flags}
mkrzero.build.ldscript=linker_scripts/gcc/flash_with_bootloader.ld
mkrzero.build.openocdscript=openocd_scripts/arduino_zero.cfg
mkrzero.build.variant=mkrzero
mkrzero.build.vid=0x2341
mkrzero.build.pid=0x804f
mkrzero.bootloader.tool=openocd
mkrzero.bootloader.file=mkrzero/samd21_sam_ba_arduino_mkrzero.bin
```

(notice the `debug` and `openocdscript` entries)

the block defining other boards such as the M0 (which has been discontinued way before we began this project and is not for sale anymore) does not have it.

If you go and edit the `M0` block you'll be able to make it work, just takes some copy/paste and trial/error, which are great activities for the weekend :) 

IMPORTANT!!!
when you update a platform these changes are lost

let me know if you manage to make it work ✌️

tuxedo0801, 2020-09-27:
Thanks @ubidefeo for your input.
I tried adding my own variant as described. The only difference I see between the EDBG and non-EDBG version of zero regarding debugging: 

```
xyz.debug.tool=gdb-openocd
```

So the new section for my "special board setup" in boards.txt now looks like:

```
# KONNEKTING SAMD Generic
# --------------------------------------
konnekting_samd_generic.name=KONNEKTING SAMD Generic
konnekting_samd_generic.vid.0=0x2341
konnekting_samd_generic.pid.0=0x804d
konnekting_samd_generic.vid.1=0x2341
konnekting_samd_generic.pid.1=0x004d
konnekting_samd_generic.vid.2=0x2341
konnekting_samd_generic.pid.2=0x824d
konnekting_samd_generic.vid.3=0x2341
konnekting_samd_generic.pid.3=0x024d

# for debugging purpose
konnekting_samd_generic.debug.tool=gdb-openocd

konnekting_samd_generic.upload.tool=bossac
konnekting_samd_generic.upload.protocol=sam-ba
konnekting_samd_generic.upload.maximum_size=262144
konnekting_samd_generic.upload.maximum_data_size=32768
konnekting_samd_generic.upload.use_1200bps_touch=true
konnekting_samd_generic.upload.wait_for_upload_port=true
konnekting_samd_generic.upload.native_usb=true
konnekting_samd_generic.build.mcu=cortex-m0plus
konnekting_samd_generic.build.f_cpu=48000000L
konnekting_samd_generic.build.usb_product="Arduino Zero"
konnekting_samd_generic.build.usb_manufacturer="Arduino LLC"
konnekting_samd_generic.build.board=SAMD_ZERO
konnekting_samd_generic.build.core=arduino
konnekting_samd_generic.build.extra_flags=-D__SAMD21G18A__ {build.usb_flags}
konnekting_samd_generic.build.ldscript=linker_scripts/gcc/flash_with_bootloader.ld
konnekting_samd_generic.build.openocdscript=openocd_scripts/arduino_zero.cfg
konnekting_samd_generic.build.variant=arduino_zero
konnekting_samd_generic.build.variant_system_lib=
konnekting_samd_generic.build.vid=0x2341
konnekting_samd_generic.build.pid=0x804d
konnekting_samd_generic.bootloader.tool=openocd
konnekting_samd_generic.bootloader.file=zero/samd21_sam_ba.bin
```
As I use the ATMEL ICE debugger, I need to tell the Arduino Pro IDE the debugu launch configuration. I had this before in VS Code working. The launch.json looks like this:

```json
{
    "version": "0.2.0",
    "configurations": [                  
        {
            "name": "Atmel ICE - SAMD",
            "type": "arduino",
            "request": "launch",
            "program": "${file}",
            "cwd": "${workspaceFolder}",
            "MIMode": "gdb",
            "targetArchitecture": "arm",
            "miDebuggerPath": "/home/alexander/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-gdb",
            "debugServerPath": "/home/alexander/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino5/bin/openocd",
            "debugServerArgs": " -d2 -s /home/alexander/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino5/share/openocd/scripts/ -f /home/achristian/.arduino15/packages/arduino/hardware/samd/1.8.8/variants/arduino_zero/openocd_scripts/arduino_zero.cfg",
            "customLaunchSetupCommands": [
                {
                    "text": "target remote localhost:3333"
                },
                {
                    "text": "file \"${file}\""
                },
                {
                    "text": "load"
                },
                {
                    "text": "monitor reset halt"
                },
                {
                    "text": "monitor reset init"
                }
            ],
            "stopAtEntry": true,
            "serverStarted": "Info\\ :\\ [\\w\\d\\.]*:\\ hardware",
            "launchCompleteCommand": "exec-continue",
            "filterStderr": true,
            "args": [],
            "logging": { "trace": true, "traceResponse": true, "engineLogging": true }
        }
    ]
}
```
I no longer receive the message with missing debug.tool property. It now tells me: 

`The gdb debugger terminated unexpectedly.`

... without any further details. Any hints on what's still wrong?

ubidefeo, 2020-09-28:
hi @tuxedo0801 

in theory once you change those files the Pro IDE should pick them up and let you debug.
@per1234 I am missing something?

tuxedo0801, 2020-09-28:
I checked the paths and files mentioned in launch.json twice... all fine, all existing, no typo.

Looks like there is another issue invoking gdb. Is there a log file I can check or log-level I can activate?

br,
Alex

ubidefeo, 2020-09-28:
you could run the debug in the CLI and add a `-v`, but not sure what that would yield.
I'll have to run a test myself with your entries

get back to you when I'm done :)

ubidefeo, 2020-09-28:
@tuxedo0801 
I don't have an M0, so I targeted a MKR1000 as your fictional board and it works for me.
You need to first compile or upload the sketch, in order for the script to find what it needs (`.elf` with debugging symbols).

Unless the M0 (which is a board we retired a while back) has some other weird quirks it should work.
It just requires the SWD header to be correctly wired.

do you have all the latest platforms updated through the Boards Manager?
[image]

tuxedo0801, 2020-09-30:
I finally set the board to MKR1000 (which is more or less compatible to mine) and now debugging works fine.

Now I have to found out what's wrong with my own board definition. Thanks so far @ubidefeo

ubidefeo, 2020-09-30:
it's my pleasure, @tuxedo0801 
we're hear to help :)

tuxedo0801, 2020-10-01:
Would be great if IDE could tell the user, that Debugging is not foreseen for the selected board, instead of failing with `undefined 'debug.tool' property`.

Then, the issue can be closed.

ubidefeo, 2020-10-01:
@tuxedo0801 
you have a really good point there.
I'll put it in our list to do asap :)