arduino/arduino-pro-ide #220: Please add a multi cursor selection, similar to what we see on Sublime Text.

RaiyaanAbdullah, 2020-02-29:
While using Arduino, we sometimes have to edit the same thing on multiple lines. If we could press Ctrl+Left Mouseclick to place multiple cursors on multiple lines and edit them all at once, it would boost efficiency. Text editors like Sublime text already have this feature.

Master811129, 2020-02-29:
> While using Arduino, we sometimes have to edit the same thing on multiple lines. If we could press Ctrl+Left Mouseclick to place multiple cursors on multiple lines and edit them all at once, it would boost efficiency. Text editors like Sublime text already have this feature.

You can use alt + left mouse click

closed 2020-02-29 by RaiyaanAbdullah

Master811129, 2020-02-29:
> Thanks.

Anytime 😉