arduino/arduino-pro-ide #224: library auto code a

lucasviniciu, 2020-03-02:
hello arduino friends, I love the pro ide but like him and pro... go there you can put a COMPLETE library in the arduino that detects in the code that does not have in the ide the library sweet potato.h then it lowers 
and my tip      @love

ubidefeo, 2020-03-03:
@masci @cmaglie @rsora 
my brain is hurting. is this spam?

lucasviniciu, 2020-03-04:
man I speak portuguese my translation and get fucked anyway LOL @Mw

Obter o Outlook para iOS<https://aka.ms/o0ukef>

per1234, 2021-02-18:
Hi @lucasviniciu . Thanks so much for taking the time to report this!

This should now be working in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4).

If you try it out and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-18 by per1234