arduino/arduino-pro-ide #225: Support pausing a debug session on Windows

spoenemann, 2020-03-04:
**Is your feature request related to a problem? Please describe.**
Currently it is not possible to pause / interrupt a debug session on Windows (breakpoints work, but you cannot use the pause button).

**Additional context**
`gdb` expects a `SIGINT` signal to interrupt a session. Windows does not support Unix signals.

spoenemann, 2020-03-05:
This is currently a problem. If someone has a hint how to interrupt a GDB session started via `arduino-cli debug` on Windows, please comment here.

To experiment:
```
arduino-cli debug -p none -b arduino:samd:arduino_zero_edbg my_sketch
```
then enter `-exec-continue` to run the program.

per1234, 2021-02-18:
The debugger now supports pause on Windows perfectly.

closed 2021-02-18 by per1234