arduino/arduino-pro-ide #226: Feature request - Fold all functions

jkenny23, 2020-03-05:
**Is your feature request related to a problem? Please describe.**
Other IDEs/editors have a feature to fold/collapse all functions, making code navigation easier. 

**Describe the solution you'd like**
Please implement the "fold all" feature which you can see in Notepad++ as an example:
https://imgur.com/WxDbkIR

**Describe alternatives you've considered**
Outline feature mostly covers this feature request, but for a "pro" IDE, it should be available as an option.

george-vieira, 2020-03-05:
+1 Yes! I consider that a fundamental of editor functions.

spoenemann, 2020-03-09:
This is already available, it's the same shortcut as in [VS Code](https://code.visualstudio.com/docs/getstarted/keybindings) (see `editor.foldAll`).

Did you know you can search for commands by pressing `F1`? (same as in VS Code)

closed 2020-03-09 by spoenemann

george-vieira, 2020-03-21:
https://code.visualstudio.com/shortcuts/keyboard-shortcuts-windows.pdf
yeah not quite the same according to these shortcuts. and the F1 menu controls eg CTRL+K or CTRL+0  doesn't do anything for me.