arduino/arduino-pro-ide #228: Failed to find "Export Compiled Binary Arduino" option

umerao, 2020-03-13:
I tried to find out “export compiled binary arduino” option which is also called .hex file generation but I failed to find out in Arduino PRO IDE. Which is easily available in standard Arduino IDE. Please tell me that is it available or not in Arduino PRO IDE?

Master811129, 2020-03-13:
> I tried to find out “export compiled binary arduino” option which is also called .hex file generation but I failed to find out in Arduino PRO IDE. Which is easily available in standard Arduino IDE. Please tell me that is it available or not in Arduino PRO IDE?

If you compile a sketch it will automatically generate a .hex file.

xdjpcm, 2020-03-14:
Also, I didn't find the option to export the binary file needed to upload to the aaruino zero board and the subsequent gebug.
In short, the debug on the samds board under windows does not work yet.
Paul

per1234, 2020-03-14:
@xdjpcm it works fine for me with SAMD boards and Windows. You don't need to do anything special with uploading the binary file. Just use the standard Upload feature.

Try this:

1. Plug the USB cable into the Programming Port of your Zero.
2. Connect the USB cable to your computer.
3. Open the sketch you want to debug.
4. Select "Arduino Zero (Programming Port)" from the board menu.
5. **Sketch > Upload**
6. Wait for the upload to finish.
7. Set breakpoints in the sketch (you can do this by clicking on the left margin of the sketch window or via the Debug menu).
8. **Debug > Start Debugging**
9. The debugging session should now start. You can control execution and see debugging information in the panel on the left side of the screen.

xdjpcm, 2020-03-14:
Thanks for the reply,
but that's not how it works. I have alpha 5 installed.
Arduino M0 Pro on COM2 port programming.
The console writes:
Error: couldn't open d: \ arduino sketch \ attempt / try.arduino.samd
Embedded: startup.tcl: 477: Error: ** Programming Failed **

The bin file is not actually in the folder, so you cannot program it.

Thanks, Paul

per1234, 2020-03-14:
OK, you said before you were using "aaruino zero board", not the Arduino M0 Pro. Debugging in Arduino Pro IDE is not supported for that board. If you use the Arduino Zero or the MKR or Nano 33 IoT boards (with an external CMSIS-DAP compatible debugger), you'll be able to debug.

xdjpcm, 2020-03-15:
Thank you very much for answering the questions.
I will try to get a board according to your recommendation and also a debugger.
Thank you once again.

per1234, 2020-03-15:
You're welcome. The [Arduino Zero](https://store.arduino.cc/arduino-zero) is nice because it has a built-in debugger. This provides the easiest road to getting started with using the debugging features of the Arduino Pro IDE.

The second easiest is the MKR1000. The reason is this board comes with the standard 0.05" pitch 2x5 male SWD header that most debuggers can plug in to. It doesn't have a built-in debugger, so you need to buy one separately. Here is a reasonably priced, open source debugger you can use with the Arduino Pro IDE:
https://www.tindie.com/products/ataradov/cmsis-dap-compliant-swd-debugger

There are plenty of other good options, that's just the one I have experience with.

You can use the same debugger with the other MKR boards and the Nano 33 IoT. The only tricky thing is those boards don't have the header installed to connect the debugger to. The other MKR boards have the SWD pins broken out to the footprint for a 0.1" 2x3 SMD header on the bottom of the board:
[image]
Those headers are widely available and it should be an easy soldering job putting it on. After that, you'd need some sort of adapter from the 0.05" pitch SWD header on the debugger to the 0.1" pitch header on the MKR board. Something like [this ](https://www.ebay.com/itm/324024687432)would work. Or you could just solder wires straight to the pads on the board.

The Nano 33 IoT has the SWD pins broken out to test pads on the bottom of the board:
[image]
You could solder wires to them (you only really need the SWDIO and SWCLK pads, since the other three are broken out to the standard pins on the Nano 33 IoT), or you could make a cool pogo pin jig like [this](https://www.instagram.com/p/B9eTOlcnSQP/).

xdjpcm, 2020-03-15:
Once again thank you for the tutorial, I think there are more people who like to learn.
I still have the same thing about the arduino M0 pro with a built-in debugger. According to the scheme https://www.arduino.cc/en/uploads/Main/arduino-M0-pro-schematic.pdf the board has debugger pins on header. The R2 and R3 resistors are used to disconnect the processor.
Do you think it could work?
Still note, according to the Arduino M0 pro and Arduino zero have the same processors, wouldn't M0 pro replace the Arduino zero bootloader? Thanks for the answers.

Paul

per1234, 2020-03-16:
I'm sure that it's somehow possible to do debugging on the M0 Pro. Unfortunately, I don't own this board and have no experience using it.

All I know is that when the debugging support was added to the Arduino SAMD Boards platform, the M0 Pro was skipped over:
https://github.com/arduino/ArduinoCore-samd/compare/d78ab37d997e610440bfcede2ca2e777a964b39d...f852a0836b6f846b6d88e1413417fde6944cc82e
This is what I meant when I said "Debugging in Arduino Pro IDE is not supported for that board".

It may be that you can simply make the modifications to the M0 Pro board definition in the Arduino SAMD Boards boards.txt equivalent to what was done for the other boards:
```
mzero_pro_bl_dbg.debug.tool=gdb
```

xdjpcm, 2020-03-16:
Thanks for the answers,
I tried to modify the boards.txt file, but it didn't work.
Then I tried in boards.txt to modify the vid and pid boards to report as arduino zero, it didn't work either.
I switched the board from M0 pro to zero, voices not connected, recompile and debugger can be programmed.
Once again, thank you very much for your time and quick answers.
Paul

per1234, 2021-02-18:
[The new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4) does have **Sketch > Export Binary**, just like the classic Arduino IDE, so I'll close this as resolved.

One difference to note: the exported binary is now saved under the `build` subfolder of the sketch, in a folder named according to the FQBN of the selected board. This is the same behavior used in [Arduino CLI](https://github.com/arduino/arduino-cli).

closed 2021-02-18 by per1234