arduino/arduino-pro-ide #232: Board search should be fuzzy

kittaakos, 2020-03-21:
Searching for `ar du` in the _Board Select_ should return with result `Arduino Duemilanove or Diecimila`. Current behavior: no match.

kittaakos, 2020-03-27:
It will be available as an experimental feature in  0.0.6.

noob-master147, 2020-04-03:
Hey, @kittaakos 
I would like to work on this bug

kittaakos, 2020-04-03:
Thank you for your interest, @noob-master147. Unfortunately, the Pro IDE code is not open-sourced yet 😕

And we have this new feature kind of available from the next release:

[image]

Did you know; the [Arduino CLI](https://github.com/arduino/arduino-cli) powers the Pro IDE. The CLI code is open source, if you want to  contribute there, the is an open feature request for the fuzzy search: https://github.com/arduino/arduino-cli/issues/629

You will need some golang knowledge ;)

noob-master147, 2020-04-03:
thanks @kittaakos 
i will surely check that out

kittaakos, 2020-04-28:
Now it works even with whitespces.

[image]

Master811129, 2020-04-29:
> Now it works even with whitespces.
> 
> [image]

nice!
does it also search 3rd party cores?

kittaakos, 2020-04-29:
Sure.

[image]

PS: I will ask around what's holding back the release.

closed 2020-05-05 by kittaakos