arduino/arduino-pro-ide #236: Upload problem on Ubuntu 19.10

leogaggl, 2020-03-31:
Unable to upload from Arduino Pro IDE on Ubuntu 19.10. 

[INFO] Searching tools required for board arduino:samd:mkrwan1310
[INFO] Required tool [tool: {ToolName:"CMSIS",ToolVersion:"4.5.0",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"CMSIS-Atmel",ToolVersion:"1.2.0",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"arduinoOTA",ToolVersion:"1.2.1",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"arm-none-eabi-gcc",ToolVersion:"7-2017q4",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"bossac",ToolVersion:"1.7.0-arduino3",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"openocd",ToolVersion:"0.10.0-arduino7",ToolPackager:"arduino"}]
[INFO] Adding libraries dir [dir: /home/USERID/software/arduino-1.8.11/libraries, location: ide]
[INFO] Adding libraries dir [dir: /home/USERID/.arduino15/packages/arduino/hardware/samd/1.8.6/libraries, location: platform]
[INFO] Adding libraries dir [dir: /home/USERID/Arduino/libraries, location: user]
[INFO] Resolving include MKRWAN.h for arch samd
[INFO]   found better lib [lib: MKRWAN, prio: 5E9]
[INFO] current best: arduino:samd@1.8.6 [bundle: false, managed: true, version: 1.8.6]
[INFO] Searching tools required for board arduino:samd:mkrwan1310
[INFO] Required tool [tool: {ToolName:"CMSIS",ToolVersion:"4.5.0",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"CMSIS-Atmel",ToolVersion:"1.2.0",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"arduinoOTA",ToolVersion:"1.2.1",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"arm-none-eabi-gcc",ToolVersion:"7-2017q4",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"bossac",ToolVersion:"1.7.0-arduino3",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"openocd",ToolVersion:"0.10.0-arduino7",ToolPackager:"arduino"}]
[INFO] Tool required for upload [tool: Version: 4.5.0 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: all]]
[INFO] Tool required for upload [tool: Version: 1.2.0 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: all]]
[INFO] Tool required for upload [tool: Version: 1.2.1 Systems: [OS: i686-linux-gnu, OS: x86_64-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: i686-mingw32]]
[INFO] Tool required for upload [tool: Version: 7-2017q4 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu]]
[INFO] Tool required for upload [tool: Version: 1.7.0-arduino3 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu]]
[INFO] Tool required for upload [tool: Version: 0.10.0-arduino7 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: x86_64-linux-gnu, OS: i686-linux-gnu, OS: i686-mingw32]]
[INFO] Tool required for upload [tool: Version: 5.8-arduino11]
[INFO] Tool required for upload [tool: Version: 1.0.0 Systems: [OS: i686-pc-linux-gnu, OS: x86_64-pc-linux-gnu, OS: i686-mingw32, OS: x86_64-apple-darwin, OS: arm-linux-gnueabihf, OS: arm64-linux-gnueabihf]]
[INFO] Touching port /dev/ttyACM0 at 1200bps
[INFO] Waiting for upload port...
[INFO] "No new serial port detected."
[INFO] current best: arduino:samd@1.8.6 [bundle: false, managed: true, version: 1.8.6]
[INFO] current best: arduino:samd@1.8.6 [bundle: false, managed: true, version: 1.8.6]

[image]

The upload, however, works without problems using arduino-cli:

>arduino-cli upload --port /dev/ttyACM0 --fqbn arduino:samd:mkrwan1310 sketch_march31b
Atmel SMART device 0x10010005 found
Device       : ATSAMD21G18A
Chip ID      : 10010005
Version      : v2.0 [Arduino:XYZ] Apr 30 2019 12:41:57
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
Boot Flash   : true
BOD          : false
BOR          : false
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
done in 0.852 seconds

Write 19048 bytes to flash (298 pages)
[==============================] 100% (298/298 pages)
done in 0.114 seconds

Verify 19048 bytes of flash with checksum.
Verify successful
done in 0.017 seconds
CPU reset.

**SYSTEM Info**
 - OS: Ubuntu 19.10 64bit
 - arduino-pro-ide 0.0.5
 - arduino-cli Version: 0.9.0 Commit: e4d02b4

kittaakos, 2020-04-06:
@leogaggl, the warning message `Reconnecting Arduino MKR WAN1310 in x seconds...` comes from the monitor feature. It tries to reconnect; I suspect, you have tried to upload to the board whilst the serial monitor was connected. The two cannot work in parallel.  Can you please try to toggle/hide the serial-monitor widget, it should automatically disconnect from the board, then try the upload again?

per1234, 2021-02-18:
>Can you please try to toggle/hide the serial-monitor widget, it should automatically disconnect from the board, then try the upload again?

This is now handled automatically (awesome!), so I'll go ahead and close this as resolved.

closed 2021-02-18 by per1234