arduino/arduino-pro-ide #238: Standard Libraries unavailable

Oldman1066, 2020-04-02:
**Describe the bug**
Importing standard libraries is impossible. When the library manager comes up the standard libraries are not there i.e. standard Servo so that basic servo function can be used Servo.h so that Servo myservo can be used.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 Windows 10

**Additional context**
Add any other context about the problem here.

per1234, 2020-04-02:
>When the library manager comes up the standard libraries are not there i.e. standard Servo

Are you saying that you can't find the Servo library listed in Library Manager? Or are you saying that you are expecting the Servo library to be pre-installed?

[image]

Oldman1066, 2020-04-02:
So sorry I missed that one completely.
Than you so much.

closed 2020-04-06 by mastrolinux