arduino/arduino-pro-ide #239: Anyone know how to add the ESP8266 boards in PRO IDE

mikeboski, 2020-04-06:
I know how to do it in the standard Arduino IDE.
Thanks for any help this software looks awesome.

kittaakos, 2020-04-07:
This feature will be available with the next release, I am still working on the finalization. See: #177 and #215.

mikeboski, 2020-04-07:
You are awesome

Hennadii65, 2020-04-09:
It is advisable to add the Developer board Robotdyn Mega2560+ESP8266. Best regards!

mikeboski, 2020-04-09:
And that will let me use the Wemos ESP12 D1 boards?
Thanks for your help.
How do I add boards I couldn't figure it out. 
Thanks for your time and kindness

Master811129, 2020-04-12:
> This feature will be available with the next release, I am still working on the finalization. See: #177 and #215.

we can't wait

brandanop, 2020-05-06:
You have to fo to File... Settings... Open CLI Configuration... and then update the config to add the new board link as shown:

board_manager:
  additional_urls: [### http://arduino.esp8266.com/stable/package_esp8266com_index.json]

kittaakos, 2020-05-06:
I have created and pinned and issue describing the CLI config a with a bit more details: https://github.com/arduino/arduino-pro-ide/issues/245

closed 2020-07-22 by kittaakos