arduino/arduino-pro-ide #240: [0.0.5] Linux, "No Ports Discovered"

FlailAway, 2020-04-22:
**Describe the bug**
No Ports Discovered

**To Reproduce**
Linux CLI
CD to Pro folder
arduino-pro-ide and launches fine
Select Board drop down
Only option is "Select Other Board & Port"
Opens window, 
  SEARCH BOARD (EMPTY)
  PORTS "NO PORTS DISCOVERED"

**Expected behavior**
See a list of attached boards, 2-Mega, one Uno.

**Desktop (please complete the following information):**
Mint 19.3 + Cinnamon all up to date

**Additional context**
If I load 1.8.12 IDE, all 3 Ports and Boards show up just fine.

kittaakos, 2020-04-27:
> NO Internet connection on this PC

Nice catch, thank you for reporting, @FlailAway. This seems to be the root of all issues, I am going to verify it and if this is an issue, the Pro IDE must handle no Internet connection gracefully.

kittaakos, 2020-04-27:
> I am going to verify it

Confirmed. The CLI can not perform the `arduino-cli core update-index` command:
```
daemon INFO Checking if CLI is Bundled into the IDE
daemon INFO Loading hardware from: /Users/akos.kitta/Library/Arduino15/packages
daemon INFO Adding libraries dir [dir: /Users/akos.kitta/Documents/Arduino/libraries, location: user]
root ERROR Uncaught Exception:  Error: 2 UNKNOWN: downloading builtin:ctags@5.8-arduino11 tool: Get https://downloads.arduino.cc/tools/ctags-5.8-arduino11-pm-x86_64-apple-darwin.zip: dial tcp: lookup downloads.arduino.cc: no such host
root ERROR Error: 2 UNKNOWN: downloading builtin:ctags@5.8-arduino11 tool: Get https://downloads.arduino.cc/tools/ctags-5.8-arduino11-pm-x86_64-apple-darwin.zip: dial tcp: lookup downloads.arduino.cc: no such host
    at callErrorFromStatus (/Applications/Arduino Pro IDE_0.0.5.app/Contents/Resources/app/node_modules/@grpc/grpc-js/build/src/call.js:30:26)
    at Http2CallStream.call.on (/Applications/Arduino Pro IDE_0.0.5.app/Contents/Resources/app/node_modules/@grpc/grpc-js/build/src/call.js:79:34)
    at Http2CallStream.emit (events.js:187:15)
    at process.nextTick (/Applications/Arduino Pro IDE_0.0.5.app/Contents/Resources/app/node_modules/@grpc/grpc-js/build/src/call-stream.js:100:22)
    at process._tickCallback (internal/process/next_tick.js:61:11)
```

We have to handle this gracefully,  and report the issue to the user. Obviously, it cannot work without an Internet connection.

kittaakos, 2020-04-27:
The only thing we can do is to warn the user when there is no Internet connection. The Pro IDE behavior remains indeterministic without a connection.

[image]

closed 2020-04-27 by kittaakos