arduino/arduino-pro-ide #242: Kernel panic

DeeEmm, 2020-04-24:
**Describe the bug**
Mac goes into Kernel Panic and reboots.

**To Reproduce**
Steps to reproduce the behavior:
- Have not been able to identify trigger for action. 
- Has happened on a number of occasions both when using the app and also when app has been in focus but doing nothing

**Expected behavior**
Kernel Panic

**Panic report**

```
Anonymous UUID:       C76CC140-418E-8598-0B62-8E57A13DBB1A

Fri Apr 24 02:14:40 2020

*** Panic Report ***
panic(cpu 3 caller 0xffffff8022ebb45d): "a freed zone element has been modified in zone kalloc.32: expected 0xe4988e2bbdaa42aa but found 0xffa440, bits changed 0xe4988e2bbd55e6ea, at offset 0 of 32 in element 0xffffff80668e2860, cookies 0x3f00110c960f5580 0x53521c8f508a9c5"@/BuildRoot/Library/Caches/com.apple.xbs/Sources/xnu/xnu-4570.71.73/osfmk/kern/zalloc.c:1122
Backtrace (CPU 3), Frame : Return Address
0xffffff820c8c3720 : 0xffffff8022e6ce56 
0xffffff820c8c3770 : 0xffffff8022f96434 
0xffffff820c8c37b0 : 0xffffff8022f88604 
0xffffff820c8c3820 : 0xffffff8022e1ee60 
0xffffff820c8c3840 : 0xffffff8022e6c8cc 
0xffffff820c8c3970 : 0xffffff8022e6c68c 
0xffffff820c8c39d0 : 0xffffff8022ebb45d 
0xffffff820c8c3a30 : 0xffffff8022ebad30 
0xffffff820c8c3a80 : 0xffffff8022eb9733 
0xffffff820c8c3bb0 : 0xffffff8022e78894 
0xffffff820c8c3c20 : 0xffffff8022e78ab1 
0xffffff820c8c3c40 : 0xffffff7fa4c5302c 
0xffffff820c8c3c80 : 0xffffff7fa4c5fbb0 
0xffffff820c8c3d60 : 0xffffff80230cd9ea 
0xffffff820c8c3dd0 : 0xffffff80230bff67 
0xffffff820c8c3e40 : 0xffffff802334684a 
0xffffff820c8c3ee0 : 0xffffff80233466ba 
0xffffff820c8c3f40 : 0xffffff8023403f48 
0xffffff820c8c3fa0 : 0xffffff8022e1f646 
      Kernel Extensions in backtrace:
         com.apple.filesystems.hfs.kext(407.50.6)[09445723-F283-31E5-B409-43D905BA81C6]@0xffffff7fa4c40000->0xffffff7fa4ca6fff
            dependency: com.apple.filesystems.hfs.encodings.kext(1)[F737F125-223E-3B67-B4FF-5AB56291D4FE]@0xffffff7fa4c38000

BSD process name corresponding to current thread: ReportCrash

Mac OS version:
17G12034

Kernel version:
Darwin Kernel Version 17.7.0: Tue Feb 18 22:51:29 PST 2020; root:xnu-4570.71.73~1/RELEASE_X86_64
Kernel UUID: F9494C59-371A-38C9-BA91-8A8AA7582B96
Kernel slide:     0x0000000022c00000
Kernel text base: 0xffffff8022e00000
__HIB  text base: 0xffffff8022d00000
System model name: iMac12,2 (Mac-942B59F58194171B)

System uptime in nanoseconds: 16125866593465
last loaded kext at 687679539711: com.apple.filesystems.msdosfs	1.10 (addr 0xffffff7fa3c3f000, size 69632)
last unloaded kext at 1091795067158: com.apple.filesystems.msdosfs	1.10 (addr 0xffffff7fa3c3f000, size 61440)
loaded kexts:
org.virtualbox.kext.VBoxNetAdp	5.2.18
org.virtualbox.kext.VBoxNetFlt	5.2.18
org.virtualbox.kext.VBoxUSB	5.2.18
com.wch.usbserial	1
org.virtualbox.kext.VBoxDrv	5.2.18
com.Cycling74.driver.Soundflower	1.6.7
com.apple.driver.AppleHWSensor	1.9.5d0
com.apple.driver.AppleTopCaseHIDEventDriver	135
com.apple.driver.AGPM	110.23.37
[... 162 lines cut ...]
Serial ATA Device: OPTIARC DVD RW AD-5690H
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: USB2.1 Hub
USB Device: CTE-450
USB Device: USB2.0-Serial
USB Device: IR Receiver
USB Device: Internal Memory Card Reader
USB Device: USB 2.0 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: Logitech Dual Action
USB Device: BRCM2046 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: iMac, Apple Inc., 25.1
```

**Desktop (please complete the following information):**
 - Osx
 - Version 10.13.6

**Additional context**
Additional possibly related behaviour is app locking up, usually with yellow / orange status bar and the close tab icons turning into a solid circular icon. This happens randomly and does not seem to be associated with any specific behaviour. The app stops responding and the only way to gain control again is to restart the app. When I close the app I get a javascript error: KILL ESRCH (see screenshot below)

[image]

DeeEmm, 2020-04-24:
Updated original post to add IDE version.

DeeEmm, 2020-04-24:
Updated original post to include App status

kittaakos, 2020-04-27:
Thank you for the detailed bug report, @DeeEmm.

The `KILL ESRCH` issue has been fixed via https://github.com/eclipse-theia/theia/issues/7113
I would expect this issue to be resolved with the next, `0.0.6-alpha` release.

DeeEmm, 2020-05-07:
Still having the crash issue on 0.0.6-alpha, but they are less frequent. 

KILL ESRCH issues has gone

kittaakos, 2020-05-07:
Thanks for the update, @DeeEmm 👍 

Just randomly googling for `a freed zone element has been modified in zone` error message, I found [this](https://apple.stackexchange.com/questions/280562/whats-causing-kernel-panic-on-mid-2011-imac-a-freed-zone-element-has-been-mod) and [this](https://discussions.apple.com/thread/8225968) threads. If you happen to have some time, could you please run a [RAM test](https://support.apple.com/en-us/HT202731) on your mac and get us back with an update? Thank you!

DeeEmm, 2020-05-08:
Will run one tonight.

DeeEmm, 2020-05-25:
Sorry for the delay, diagnostic test comes back fine.

DeeEmm, 2020-06-29:
I've migrated to a new mac. The issue is no longer present. I can only assume that the issue was related to the environment on the old mac. 

As I can no longer recreate the error, I'll close this off.

closed 2020-06-29 by DeeEmm

kittaakos, 2020-06-29:
Ohh, thank you for coming back and letting us know. 👍

DeeEmm, 2020-06-30:
No dramas. Keep up the good work. Really digging the new IDE.