arduino/arduino-pro-ide #243: Boards Manage & Manage Libraries tabs have no "close"

awgrover, 2020-04-28:
**Describe the bug**
Expected a "close" ("x") on the Boards Manager and Manage Libraries tab.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'tools' menu
2. Click on 'Boards Manager'
3. Note "Boards Manager" tab, no "x". Compare with the "ino" tab with "x".

**Expected behavior**
A "close" for the tab. Cf. the "ino" tab with "x".

A work-around is to choose "Tools":"Boards Manager" again.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04.4
 - Version arduino-pro-ide 0.0.5

robonxt, 2020-04-30:
Same issue, however once you switch to Advanced Mode, the "close" ("x") shows up properly on the tabs.

per1234, 2021-02-17:
Boards and Library Manager no longer use tabs. Instead, they are shown in the side bar. The side bar still doesn't have an "x" button, but I don't think that is a common convention for this type of GUI element.

So I'll go ahead and close this. Thanks to you both for taking the time to report this bug.

closed 2021-02-17 by per1234