arduino/arduino-pro-ide #244: Some syntax are not identified and marked as Problem

mishracodes, 2020-05-06:
**Describe the bug**
Some of the Syntax and identifiers are not recognized by editor and red underline comes under it. Like in pinMode(2,OUTPUT); here OUTPUT is not recognized and says "**Use of undeclared identifier 'OUTPUT'clang(undeclared_var_use)**"
There are many more like
**Serial** in Serial.begin();
**A0**  port number 
 **delay** in delay(1000);
**LED_BUILTIN** 
**LOW** in digitalWrite(2,LOW);
**HIGH**  in digitalWrite(2,HIGH);
**digitalRead** but not digitalWrite and many more
The code is verified successfully and uploading also but only showing underline in editor and also in the **Problem** panel.

Also I added Simple DHT Library and on declaring DHT object it is showing red underline but the code is working perfectly and giving reading on the serial monitor.

**Expected behavior**
It should not display error and problems in the system defined functions and constants and also in other functions, constants and objects if its library is included.

**Screenshots**
https://drive.google.com/open?id=1DaicDwp4aAcbiV7KlOBb-qTKdS2Z69fs

**Desktop (please complete the following information):**
 - OS: [Windows]
 - Version [10]

ThinCode2020, 2020-09-27:
**As of 09-26-2020 still having a lot of under line error:**
pinMode(buttonPin, INPUT);   Use of undeclared identifier 'INPUT' clang(undeclared_var_use)
pinMode(ledPin, OUTPUT);   Use of undeclared identifier 'OUTPUT' clang(undeclared_var_use)
Serial.begin(9600);  Use of undeclared identifier 'Serial' clang(undeclared_var_use)
buttonState = digitalRead(buttonPin);    Use of undeclared identifier.....
if (buttonState == HIGH)       Use of undeclared identifier.....
delay(50);       Use of undeclared identifier.....
 digitalWrite(ledPin, HIGH);      Use of undeclared identifier.....
and  more.

Please fix

ubidefeo, 2020-09-27:
hi @ThinCode2020 
The language server is undergoing an extensive rework, hence we'll keep having this kind of issues for a bit longer.
Please bear with us, we'll do it right asap 🙏

ThinCode2020, 2020-09-27:
Hello,

Thank you

per1234, 2021-02-17:
Hi @mishracodes. Thanks so much for taking the time to report this!

This should now be working in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4).

If you try it out and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-17 by per1234