arduino/arduino-pro-ide #249: Command + R shortcut not working in other apps while the IDE is running

insidegui, 2020-05-08:
**Describe the bug**
On my Mac, while the IDE is running, the ⌘R keyboard shortcut stops working in other apps such as Safari and Xcode.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch the IDE on the Mac
2. Try using the Command + R keyboard shortcut in another app
**Expected behavior**
The app should not take ownership of a standard keyboard shortcut such as Command + R when it's not the active app.

**Desktop (please complete the following information):**
 - OS: macOS
 - Version 10.15.4

ubidefeo, 2020-05-09:
hi @insidegui 
This is a weird behaviour, the app should not capture and hold any shortcut,
especially CMD+R which we don't use. Must be an Electron thing 😓 
I'll make sure we check this behaviour next week.

ps: big fan :)

kittaakos, 2020-05-14:
Thanks for logging it, yes I can confirm this; I was an idiot and overridden the OS level <kbd>Ctrl/⌘</kbd>+<kbd>R</kbd>.

Refs:
 - Issue: https://github.com/electron/electron/issues/3725#issuecomment-540854069
 - Better fix: https://electron.guide/final-polish/renderer/#prevent-browserwindow-refreshes

kittaakos, 2020-05-14:
@insidegui, the [fix](https://github.com/eclipse-theia/theia/pull/7817) is ready but I am afraid, all of us have to wait for the next release cycle. There is a workaround, however, one can replace the `resources/app/src-gen/frontend/electron-main.js` manually with the JS from below. 👇

The `electron-main.js` file can be found at:
 - Windows: `"C:\\path\\to\\Arduino Pro IDE-v0.0.x-win\\resources\\app\\src-gen\\frontend\\electron-main.js"`,
 - macOS: `/path/to/Arduino\ Pro\ IDE.app/Contents/Resources/app/src-gen/frontend/electron-main.js`, and
 - Linux: `/path/to/Arduino Pro IDE-v0.0.x-linux-x64/resources/app/src-gen/frontend/electron-main.js`.

Let me know if the problem persists. I am going to close this, please reopen if required.

```js
// @ts-check

// Useful for Electron/NW.js apps as GUI apps on macOS doesn't inherit the `$PATH` define
// in your dotfiles (.bashrc/.bash_profile/.zshrc/etc).
// https://github.com/electron/electron/issues/550#issuecomment-162037357
// https://github.com/eclipse-theia/theia/pull/3534#issuecomment-439689082
require('fix-path')();

// Workaround for https://github.com/electron/electron/issues/9225. Chrome has an issue where
// in certain locales (e.g. PL), image metrics are wrongly computed. We explicitly set the
// LC_NUMERIC to prevent this from happening (selects the numeric formatting category of the
// C locale, http://en.cppreference.com/w/cpp/locale/LC_categories).
if (process.env.LC_ALL) {
    process.env.LC_ALL = 'C';
}
process.env.LC_NUMERIC = 'C';

const { v4 } = require('uuid');
const electron = require('electron');
const { join, resolve } = require('path');
const { fork } = require('child_process');
const { app, dialog, shell, BrowserWindow, ipcMain, Menu, globalShortcut } = electron;
const { ElectronSecurityToken } = require('@theia/core/lib/electron-common/electron-token');

const applicationName = `Theia Electron Example`;
const isSingleInstance = false;
const disallowReloadKeybinding = false;

if (isSingleInstance && !app.requestSingleInstanceLock()) {
    // There is another instance running, exit now. The other instance will request focus.
    app.quit();
    return;
}

const nativeKeymap = require('native-keymap');
const Storage = require('electron-store');
const electronStore = new Storage();

const electronSecurityToken = {
    value: v4(),
};

app.on('ready', () => {

    // Explicitly set the app name to have better menu items on macOS. ("About", "Hide", and "Quit")
    // See: https://github.com/electron-userland/electron-builder/issues/2468
    app.setName(applicationName);

    const { screen } = electron;

    // Remove the default electron menus, waiting for the application to set its own.
    Menu.setApplicationMenu(Menu.buildFromTemplate([{
        role: 'help', submenu: [{ role: 'toggledevtools'}]
    }]));

    function createNewWindow(theUrl) {

        // We must center by hand because `browserWindow.center()` fails on multi-screen setups
        // See: https://github.com/electron/electron/issues/3490
        const { bounds } = screen.getDisplayNearestPoint(screen.getCursorScreenPoint());
[... 184 lines cut ...]
            console.error(error);
            app.exit(1);
        });
    } else {
        // We want to pass flags passed to the Electron app to the backend process.
        // Quirk: When developing from sources, we execute Electron as `electron.exe electron-main.js ...args`, but when bundled,
        // the command looks like `bundled-application.exe ...args`.
        const cp = fork(mainPath, process.argv.slice(devMode ? 2 : 1), { env: Object.assign({
            [ElectronSecurityToken]: JSON.stringify(electronSecurityToken),
        }, process.env) });
        cp.on('message', (address) => {
            loadMainWindow(address.port);
        });
        cp.on('error', (error) => {
            console.error(error);
            app.exit(1);
        });
        app.on('quit', () => {
            // If we forked the process for the clusters, we need to manually terminate it.
            // See: https://github.com/eclipse-theia/theia/issues/835
            process.kill(cp.pid);
        });
    }
});
```

closed 2020-05-14 by kittaakos

ubidefeo, 2020-05-14:
@insidegui 
if you need it right away try and patch according to @kittaakos 's directions, otherwise you'll have to wait for the next minor :)

insidegui, 2020-05-14:
Awesome! Thanks