arduino/arduino-pro-ide #250: Not working on Kubuntu 19.04 (go offline)

ricardojlrufino, 2020-05-09:
**Describe the bug**
After opening an IDE, it soon appears as OffLine in the status bar

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Kubtunu 19.04
 - Version v0.0.6-alpha.preview

**Additional context**
Console error
```
root INFO Changed application state from 'initialized_layout' to 'ready'.
daemon INFO current best: arduino:avr@1.8.2 [bundle: false, managed: true, version: 1.8.2]
daemon INFO current best: arduino:sam@1.6.11 [bundle: false, managed: true, version: 1.6.11]
daemon INFO current best: digistump:avr@1.6.7 [bundle: false, managed: true, version: 1.6.7]
daemon INFO current best: esp8266:esp8266@2.4.1 [bundle: false, managed: true, version: 2.4.1]
daemon INFO current best: Cosa:avr@1.2.0 [bundle: false, managed: true, version: 1.2.0]
daemon INFO current best: esp32:esp32@1.0.4 [bundle: false, managed: true, version: 1.0.4]
daemon INFO current best: Cosa:avr@1.2.0 [bundle: false, managed: true, version: 1.2.0]
daemon INFO current best: esp32:esp32@1.0.4 [bundle: false, managed: true, version: 1.0.4]
daemon INFO current best: arduino:avr@1.8.2 [bundle: false, managed: true, version: 1.8.2]
daemon INFO current best: arduino:sam@1.6.11 [bundle: false, managed: true, version: 1.6.11]
daemon INFO current best: digistump:avr@1.6.7 [bundle: false, managed: true, version: 1.6.7]
daemon INFO current best: esp8266:esp8266@2.4.1 [bundle: false, managed: true, version: 2.4.1]
root INFO Index has been updated.
daemon INFO panic: runtime error: invalid memory address or nil pointer dereference
daemon INFO [signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x7b926c]
daemon INFO goroutine 15 [running]:
daemon INFO go.bug.st/relaxed-semver.(*Version).CompareTo(0xc0020e2630, 0x0, 0xc00149e8c0)
daemon INFO     /github/home/go/pkg/mod/go.bug.st/relaxed-semver@v0.0.0-20190922224835-391e10178d18/version.go:115 +0x3c
daemon INFO go.bug.st/relaxed-semver.(*Version).GreaterThan(...)
daemon INFO     /github/home/go/pkg/mod/go.bug.st/relaxed-semver@v0.0.0-20190922224835-391e10178d18/version.go:202
daemon INFO github.com/arduino/arduino-cli/arduino/libraries/librariesindex.(*Index).FindLibraryUpdate(0xc00000e008, 0xc000488360, 0x0)
daemon INFO     /__w/arduino-cli/arduino-cli/arduino/libraries/librariesindex/index.go:121 +0x88
daemon INFO github.com/arduino/arduino-cli/commands/lib.listLibraries(0xc002e80340, 0xc000170000, 0x1, 0xc0000b2178, 0x1)
daemon INFO     /__w/arduino-cli/arduino-cli/commands/lib/list.go:65 +0xe8
daemon INFO github.com/arduino/arduino-cli/commands/lib.LibraryList(0xe0f800, 0xc001d94480, 0xc001d944b0, 0xcc4700, 0xc0004f8300, 0xe1dd40)
daemon INFO     /__w/arduino-cli/arduino-cli/commands/lib/list.go:38 +0xb5
daemon INFO github.com/arduino/arduino-cli/commands/daemon.(*ArduinoCoreServerImpl).LibraryList(0xc0001b0f60, 0xe0f800, 0xc001d94480, 0xc001d944b0, 0xc0001b0f60, 0xc001d94480, 0xc00009fb30)
daemon INFO     /__w/arduino-cli/arduino-cli/commands/daemon/daemon.go:288 +0x3f
daemon INFO github.com/arduino/arduino-cli/rpc/commands._ArduinoCore_LibraryList_Handler(0xcc4700, 0xc0001b0f60, 0xe0f800, 0xc001d94480, 0xc0004f8300, 0x0, 0xe0f800, 0xc001d94480, 0xc000617240, 0x4)
daemon INFO     /__w/arduino-cli/arduino-cli/rpc/commands/commands.pb.go:1799 +0x217
daemon INFO google.golang.org/grpc.(*Server).processUnaryRPC(0xc0000ae900, 0xe17e40, 0xc000001800, 0xc00112fe00, 0xc0001b3e90, 0x135ea90, 0x0, 0x0, 0x0)
daemon INFO     /github/home/go/pkg/mod/google.golang.org/grpc@v1.27.0/server.go:1024 +0x4f4
daemon INFO google.golang.org/grpc.(*Server).handleStream(0xc0000ae900, 0xe17e40, 0xc000001800, 0xc00112fe00, 0x0)
daemon INFO     /github/home/go/pkg/mod/google.golang.org/grpc@v1.27.0/server.go:1313 +0xd97
daemon INFO google.golang.org/grpc.(*Server).serveStreams.func1.1(0xc000036100, 0xc0000ae900, 0xe17e40, 0xc000001800, 0xc00112fe00)
daemon INFO     /github/home/go/pkg/mod/google.golang.org/grpc@v1.27.0/server.go:722 +0xbb
daemon INFO created by google.golang.org/grpc.(*Server).serveStreams.func1
daemon INFO     /github/home/go/pkg/mod/google.golang.org/grpc@v1.27.0/server.go:720 +0xa1
```

ricardojlrufino, 2020-05-12:
I tryied old version:0.0.5-alpha.preview-linux64/
It goes online and the Menus work. But I can't compile

I got this error:
[image]

kittaakos, 2020-05-14:
@ricardojlrufino, do you have `arduino-cli` on your `PATH`? Have you customized the CLI configuration file under `~/.arduinoProIDE/arduino-cli.yaml`?

kittaakos, 2020-05-14:
Here is a very similar error from the CLI: https://github.com/arduino/arduino-cli/issues/28
Let's keep an eye on that and reopen if required.

ricardojlrufino, 2020-05-14:
> do you have `arduino-cli` on your `PATH`? Have you customized the CLI
No

Does it matter any configuration of the old Arduino? 
For it has already opened in a random sketch ... (I think maybe even the Platform IO)

ricardojlrufino, 2020-05-14:
Hi, i fixed...

> 1. Remove : ~/.arduinoProIDE/
> 2. MOVE old: ~/.arduino15 to ~/.arduino15-bck/
> 3. MOVE: ~/Arduino

Now works...

Try revert : 2. MOVE old: ~/.arduino15-bck/ to ~/.arduino15
Works..

I think it was the folder: `~ / Arduino`, I don't know which IDE created this folder.

kittaakos, 2020-05-14:
> No

Thanks for the update.

> 1. Remove : ~/.arduinoProIDE/

This is where you store Pro IDE settings, such as your workbench layout, CLI config, other VS Code extensions, etc.

> 2\. OVE old: ~/.arduino15 to ~/.arduino15-bck/

This is managed by the CLI. I do not know much about it or why it caused any issues but we can dig deeper if you will.

> I think it was the folder: `~ / Arduino`, I don't know which IDE created this folder.

I do not know either, I bet the Java-based, "old" IDE.

> Hi, i fixed...
> Now works...

Very nice 👍

ricardojlrufino, 2020-05-14:
`~/.arduino15` is maneged by Java IDE. But as I said I moved it, then I came back and it worked ...

It is definitely something that was in the folder: /home/ricardo/Arduino
I copied the folder back that I had backed up, and the problem came up.

[image]

Probably some old library or incompatible core.
See my tree : /home/ricardo/Arduino
```
.
├── DigisparkBluetooth
│   └── DigisparkBluetooth.ino
├── ESPTcpServer
│   └── ESPTcpServer.ino
├── hardware
├── libraries
│   ├── aREST
│   │   ├── aREST.h
│   │   ├── examples
│   │   │   ├── BLE
│   │   │   │   └── BLE.ino
│   │   │   ├── ESP8266
│   │   │   │   └── ESP8266.ino
│   │   │   ├── Ethernet
│   │   │   │   └── Ethernet.ino
│   │   │   ├── Serial
│   │   │   │   └── Serial.ino
│   │   │   ├── WiFi
│   │   │   │   └── WiFi.ino
│   │   │   ├── WiFi_CC3000
│   │   │   │   └── WiFi_CC3000.ino
│   │   │   ├── WiFi_CC3000_Due
│   │   │   │   └── WiFi_CC3000_Due.ino
│   │   │   └── Yun
│   │   │       └── Yun.ino
│   │   ├── lib.index
│   │   ├── library.json
│   │   ├── library.properties
│   │   ├── README.md
│   │   └── test
│   │       ├── http_test.py
│   │       ├── lightweight_test.py
│   │       └── serial_test.py
│   ├── OpenDevice -> /media/ricardo/Dados/Codidos/Java/Projetos/OpenDevice/opendevice-hardware-libraries/arduino/OpenDevice
│   ├── readme.txt
│   └── UIPEthernet -> /media/ricardo/Dados/Codidos/Java/Projetos/OpenDevice/opendevice-hardware-libraries/arduino/UIPEthernet
├── sketch_aug09a
│   └── sketch_aug09a.ino
├── sketch_jul11b
│   └── sketch_jul11b.ino
├── sketch_sep18a
│   └── sketch_sep18a.ino
├── sketch_sep19a
│   └── sketch_sep19a.ino
├── sketch_sep19b
│   └── sketch_sep19b.ino
├── Teste1
│   └── Teste1.ino
├── Teste2
│   └── Teste2.ino
├── Teste3
│   └── Teste3.ino
├── Teste4
│   └── Teste4.ino
├── Teste5
│   └── Teste5.ino
├── Teste6
│   ├── Teste6.ino
│   └── Teste.h
├── TesteAutocomplete1
│   ├── MyClass.h
│   ├── StringUtins.ino
│   └── TesteAutocomplete1.ino
├── TesteAutocomplete2.ino
│   └── TesteAutocomplete2.ino.ino
├── TesteAutocomplete3
│   └── TesteAutocomplete3.ino
├── TesteAutocomplete4
│   └── TesteAutocomplete4.ino
├── TesteAutocomplete5
│   └── TesteAutocomplete5.ino
├── TesteAutocomplete6
│   └── TesteAutocomplete6.ino
│       └── TesteAutocomplete6.ino.ino
├── TesteAutocomplete7
│   └── TesteAutocomplete7.ino
├── TesteAutocomplete8
│   └── TesteAutocomplete8.ino
├── TesteRFID
│   └── TesteRFID.ino
└── Testeserial
    └── Testeserial.ino
```

ricardojlrufino, 2020-05-14:
Mystery solved, incompatible library.
UIPEthernet

The strange thing is that this error broke the entire IDE, nor did the menus work. It would be interesting to have an error handling in that part ..

kittaakos, 2020-05-14:
> `~/.arduino15` is maneged by Java IDE.

By the CLI, as far as I know. See the corresponding code [here](https://github.com/arduino/arduino-cli/blob/c3871677ab82229181975f6aba1170bd721a58d5/configuration/configuration.go#L90-L91). Maybe the Java IDE too, I do not know.

> Mystery solved, incompatible library.
> UIPEthernet

How did you manage to figure it out? How can I reproduce the incompatible `UIPEthernet` lib? Thanks!

> It would be interesting to have an error handling in that part ..

I agree. First, I have to understand how can I reproduce this. Can you help?

ricardojlrufino, 2020-05-14:
Test using this file please.
[UIPEthernet.zip](https://github.com/arduino/arduino-pro-ide/files/4629229/UIPEthernet.zip)

kittaakos, 2020-05-15:
@ricardojlrufino, thanks for zipping up the lib, how can I try it? Can you tell me the steps you did? I have to figure out if the CLI (without the Pro IDE) can use this lib or not.

ricardojlrufino, 2020-05-15:
Put in library folder:
/home/USER/Arduino/libraries

> have to figure out if the CLI (without the Pro IDE) can use this lib or not.
i don't using the arduino-cli

per1234, 2021-02-17:
>Test using this lib please.
UIPEthernet.zip

The problem with this library is the value of the `version` field in the library.properties metadata file is non "relaxed semver" compliant, as required by the Arduino Library specification:
```
version=1.04
```
(note the leading 0 on the minor version component)

We have a bug report (https://github.com/arduino/arduino-cli/issues/1176) and work in progress on a PR (https://github.com/arduino/arduino-cli/pull/1177) right now in the Arduino CLI repo to handle this circumstance gracefully. @kittaakos. Do you think the planned improvement in Arduino CLI will be sufficient to resolve this issue without any need for changes to Arduino Pro IDE? If so, I would go ahead and close this issue.