arduino/arduino-pro-ide #254: Could not start editor support.<br>Error during build: exec: "/bin/avr-g++": file does not exist<br>

Denontime, 2020-05-13:
**Describe the bug**
A clear and concise description of what the bug is.
when use v-0.6 on windows ,it put this ,and I can't use it
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

LumitoLuma, 2020-08-10:
@Denontime I also have this problem. I have temporary solved it (on my local computer) with this steps:

1.  Go to boards manager.
2.  Install the previous version of the board you have selected.
3.  Then, uninstall the version you have recently downloaded.
4.  Finally, install the latest version of the board.

Hope it solves your problem!

ubidefeo, 2021-02-25:
This issue can happen when the CLI fails to complete installation of a core and its tools.
It is a sporadic issue and we now have a Knowledge-Base article to address it.
https://support.arduino.cc/hc/en-us/articles/360018444739-Error-file-does-not-exist-no-such-file-or-directory-system-cannot-find-the-file-specified
This issue will hence be closed for now.
Please also test with the latest release, but it might be that you never bump into it again

closed 2021-02-25 by ubidefeo