arduino/arduino-pro-ide #255: CLI configs do not work on Windows

kittaakos, 2020-05-14:
@csaba911 wrote [here](https://github.com/arduino/arduino-pro-ide/issues/245#issuecomment-626036326):

> Awesome tutorial, much more detailed what is needed but my clean fresh install of 0.0.6 works nothing like illustrated.
> when searching for board -> not found, I mean
> [image]

kittaakos, 2020-05-14:
> clean fresh install of 0.0.6 works nothing like illustrated.

@csaba911, I could not reproduce it with the 0.0.6-alpha on Windows. Please try to start the application from a `CMD.exe`.
 - Download the ZIP and unpack,
 - `cd` to `"Arduino Pro IDE-v0.0.6-win"`,
 - start with `"Arduino Pro IDE.exe`".

Do you see any errors in the terminal? What does the status bar show, do it have a yellow (`offline`) background color? Also, please check if it isn't a [port issue on Windows](https://github.com/arduino/arduino-pro-ide/issues/134#issuecomment-553537737), please try to change the `daemon.port` config value.

per1234, 2021-02-17:
Closing due to lack of feedback. @csaba911 if you are still experiencing this issue when using [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4), and you can comment here with all the information kittaakos requested, we'll investigate further.

closed 2021-02-17 by per1234