arduino/arduino-pro-ide #256: Compile Error

atif85377, 2020-05-20:
I am getting error while compiling the program the error is as reported here:

"Error: Request compile failed with message: 2 UNKNOWN: exit status 1
"
Screenshot of the error is linked below...
[image]

atif85377, 2020-05-20:
Please help!!!

ubidefeo, 2020-05-20:
@atif85377 
please read the errors
this is the repository for this project, not the support forum for Arduino
the error you received is clearly reported in the message:
`1293_test.ino:5:1` 
which means line 5 character 1
`expected ; before }`
you forgot to end your statement.
you can see in the code window that you have a `}` symbol marked with a red underline.

questions and issues with the code are to be posted to the forum in the beginner section.
thank you
ubi

closed 2020-05-20 by ubidefeo

atif85377, 2020-05-20:
It works.
Thanks for assistance.