arduino/arduino-pro-ide #257: No syntax highlighting and intellisense for *.h and *.cpp files

theelims, 2020-05-20:
**Describe the bug**
There is no syntax highlighting and intellisense support for *.h and *.cpp files. Highlighting works fine for *.ino, however.

**To Reproduce**
Open any *.h or *.cpp file of your liking. 

**Expected behavior**
Other source files should be recognized and treated as source files by the editor.

**Screenshots**
[image]

**Desktop:**
 - OS: Windows 10 Version 1809
 - Version: 0.0.6 alpha

dejanmeznarc, 2020-05-22:
I have same issue.

THATDONFC, 2020-05-26:
I am experiencing the same thing

bustix, 2020-05-29:
Hi,

I'm experiencing the same issue unfortunately.

billsonualberta, 2020-06-06:
Same Issue Here.

I'm running MacOs Catalina (10.15.3)

Toem5, 2020-06-07:
Same issue in linux version (running Linux Mint 18.3)

closed 2020-07-03 by kittaakos

jalius, 2020-08-21:
Hi. I'm wondering if we can make a temporary workaround using the _settings.json_ of the .arduinoProIDE.
I see that we can add a preference for file associations: 
```
{
"files.associations": {"*.h":"Cpp", "*.ino":"INO"}
}
```
Is there any way to add the arduino syntax highlighting to .h and .cpp files using this? The value _Cpp_ or _Ino_ doesn't work for the language option and I can't seem to figure it out by any guesswork. It would be so helpful for using the IDE.