arduino/arduino-pro-ide #259: Undeclared identifiers for much of the arduino libs

natemara, 2020-06-02:
Many identifiers that are a part of the arduino standard library show up as unidentified in the pro IDE - here are the ones I've seen:

- `INPUT`
- `OUTPUT`
- `HIGH`
- `LOW`
- `millis`
- `digitalRead`

**To Reproduce**
Steps to reproduce the behavior:
1. Start a new sketch
2. Use one of the above identifiers
3. See red underlines and a line in the "problems" section

**Expected behavior**

These identifiers should not be underlined by the editor.

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 0.0.6

I have the Arduino AVR Boards installed in the boards manager

CharlotteDunois, 2020-06-09:
`noInterrupts` and `interrupts are also red underlined.

[image]

(using 0.0.6 alpha for Windows)

kittaakos, 2020-06-11:
You have not selected a board yet, so the underlying Arduino language server has no idea what constants are available, please select a correct board and let us know if it works.

CharlotteDunois, 2020-06-11:
@kittaakos I've selected MKR1000 board when I posted the comment. I've been compiling, uploading and debugging using Pro IDE for a few hours.

TheRijn, 2020-06-16:
Even with the boars selected I get these errors on some libs. Thinks like `EthernetClient` or `Ethernet` are not recognised. (And yes I've included `Ethernet.h`)

per1234, 2020-06-16:
>Even with the boars selected I get these errors on some libs. Thinks like EthernetClient or Ethernet are not recognised. (And yes I've included Ethernet.h)

I believe that is a separate issue: https://github.com/arduino/arduino-pro-ide/issues/274
Please check if the problem only occurs with sketches saved outside your user directory/sketchbook. You can find the location of the user directory in **File > Settings > Open CLI Configuration**.

TheRijn, 2020-06-16:
> tory/sketchbook. You can find the location of the user directory in **File > Settings > Open CLI Configuration**.

Yes, it was saved outside my user directory. This is the same issues as in #274.

per1234, 2021-02-17:
As kittaakos explained, natemara's issue was due to not having a board selected. It'required to have a board selection beffore the language server can work.

By the changed user name, I can tell that ghost has deleted their GitHub account (that is a placeholder user name automatically used by GitHub for deleted accounts), so we won't be able to investigate their report.

#274 should now be fixed in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4).

So I think everything is a wrap here and I'll go ahead and close it.

If you try out 0.1.4 and find that #274 is still occurring, feel free to comment on that issue and we'll investigate.

closed 2021-02-17 by per1234