arduino/arduino-pro-ide #262: Ghost boards appearing in board selection menu [ Mac ]

ubidefeo, 2020-06-04:
**Describe the bug**
One one of my machines when I go select a board for the current sketch, whatever I type I get two of these entries (Adafruit Grand Central M4) every time the search query changes.

**To Reproduce**
Steps to reproduce the behavior:
I have not been able to reproduce it on my other machine, but on the one I'm on now it keeps doing it.
`arduino-cli.yaml` is the same on both machines
1. Select the board dropdown and choose `Select other board  & port"
2. Type anything (I used UNO)
3. You are presented with your options plus 2 greyed out entries for `Adafruit Grande Central M4`
4. Change the search query and 2 more will add to the beginning of the list
5. This keeps on growing and active choices are pushed down

**Expected behavior**
n/a

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Mac OS Mojave
 - Version 10.14.6

**Additional context**
n/a

kittaakos, 2020-07-20:
I could reproduce this issue finally, I had to install `https://adafruit.github.io/arduino-board-index/package_adafruit_index.json` as a 3rd party core.

I immediately started to see the following error:
```
root ERROR Warning: Encountered two children with the same key, `Adafruit Grand Central M4-Adafruit SAMD Boards`. Keys should be unique so that components maintain their identity across updates. Non-unique keys may cause children to be duplicated and/or omitted — the behavior is unsupported and could change in a future version.
    in div (created by BoardsConfig)
    in div (created by BoardsConfig)
    in div (created by BoardsConfig)
    in div (created by BoardsConfig)
    in BoardsConfig
    in div
```

Something is different with this core, some IDs or names are not unique. Still checking...

kittaakos, 2020-07-20:
I can confirm, there are duplicates in the `package_adafruit_index.json` from `https://adafruit.github.io/arduino-board-index/package_adafruit_index.json`.

```json
        {
          "name": "Adafruit SAMD Boards",
          "architecture": "samd",
          "version": "1.5.2",
          "category": "Adafruit",
          "url": "https://adafruit.github.io/arduino-board-index/boards/adafruit-samd-1.5.2.tar.bz2",
          "archiveFileName": "adafruit-samd-1.5.2.tar.bz2",
          "checksum": "SHA-256:4ea84be363d82398dea03e25d1f3f5d1b40ad9658d73057749faad2a34cb2cac",
          "size": "15695783",
          "help": {
            "online": "https://forums.adafruit.com"
          },
          "boards": [
            {
              "name": "Adafruit Feather M0"
            },
            {
              "name": "Adafruit Feather M0 Express"
            },
            {
              "name": "Adafruit Metro M0 Express"
            },
            {
              "name": "Adafruit Circuit Playground Express"
            },
            {
              "name": "Adafruit Gemma M0"
            },
            {
              "name": "Adafruit Trinket M0"
            },
            {
              "name": "Adafruit ItsyBitsy M0"
            },
            {
              "name": "Adafruit pIRkey M0"
            },
            {
              "name": "Adafruit Metro M4"
            },
            {
              "name": "Adafruit Grand Central M4"
            },
            {
              "name": "Adafruit ItsyBitsy M4"
            },
            {
              "name": "Adafruit Grand Central M4"
            },
            {
              "name": "Adafruit Feather M4 Express"
            },
            {
              "name": "Adafruit Hallowing M0"
            },
            {
              "name": "Adafruit NeoTrellis M4"
            },
            {
              "name": "Adafruit PyPortal M4"
            },
            {
              "name": "Adafruit PyBadge M4"
            },
            {
              "name": "Adafruit Metro M4 AirLift"
            }
          ],
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "arm-none-eabi-gcc",
              "version": "4.8.3-2014q1"
            },
            {
              "packager": "arduino",
              "name": "bossac",
              "version": "1.7.0"
            },
            {
              "packager": "arduino",
              "name": "bossac",
              "version": "1.8.0-48-gb176eee"
            },
            {
              "packager": "arduino",
              "name": "openocd",
              "version": "0.9.0-arduino"
            },
            {
              "packager": "arduino",
              "name": "CMSIS",
              "version": "4.5.0"
            },
            {
              "packager": "arduino",
              "name": "CMSIS-Atmel",
              "version": "1.2.0"
            }
          ]
        },
```

`Adafruit Grand Central M4` is there twice.

ubidefeo, 2020-07-20:
Thank you for hunting this down, @kittaakos 
Not yet sure how to move about this, but it's definitely an issue not happening on the Java IDE, 
although the right thing to do would be to get in touch with Adafruit and have them fix this index.
@cmaglie , what do you think?

kittaakos, 2020-07-20:
> Not yet sure how to move about this,

I can propose the following; if the board is not yet installed (no FQBN) we show `Adafruit Grand Central M4` only once in boards config dialog. From the user's point of view, it does not matter whether it is there once or twice especially without the installed code. Once the corresponding core is installed, the FQBNs will be available. With the FQBNs we can distinguish between the two boards, and show both of them in the UI. What do you think?

kittaakos, 2020-07-20:
> From the user's point of view, it does not matter whether it is there once or twice especially without the installed code

This is just my naive assumption.

ladyada, 2020-07-20:
oops we can also fix it in our BSP

@hathach can you check it out and remove the duplicate & bump the samd BSP version by 0.0.1 :)

cmaglie, 2020-07-20:
> I can propose the following; if the board is not yet installed (no FQBN) we show Adafruit Grand Central M4 only once in boards config dialog. From the user's point of view, it does not matter whether it is there once or twice especially without the installed core. Once the corresponding core is installed, the FQBNs will be available. With the FQBNs we can distinguish between the two boards, and show both of them in the UI. What do you think?

:+1:

ubidefeo, 2020-07-20:
@ladyada it's a minor slip, we're happy this issue helps you do some cleanup :)

hathach, 2020-07-21:
> oops we can also fix it in our BSP
> 
> @hathach can you check it out and remove the duplicate & bump the samd BSP version by 0.0.1 :)

Done at https://github.com/adafruit/arduino-board-index/pull/70 :)

closed 2020-07-23 by kittaakos