arduino/arduino-pro-ide #263: Board search not reliable

ubidefeo, 2020-06-04:
**Describe the bug**
It appears that search won't look for string match in `board.name` but only look for the prefix boardname.

**To Reproduce**
Steps to reproduce the behavior:
1. Make sure you have `arduino:avr` core installed
2. Click on "select other board or port"
3. Type "duemila" (for Arduino Duemilanove or Diecimila)
4. No results will be presented
5. Type "dieci" (for the same boards)
6. Duemilanove or Diecimila will appear in the results

**Expected behavior**
One would expect the search to happen in the strings comprising the board name.
In the specific case look for the content of `diecimila.name=Arduino Duemilanove or Diecimila`

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Mac OS
 - Version 10.14.6

**Additional context**
n/a

ubidefeo, 2020-06-04:
Also by just typing "arduino" in the search field if you have esp8266 core installed it will show a board named "arduino" and let you select it, which will bring up an error on compile
[image]
[image]
[image]

kittaakos, 2020-07-20:
The first part of the issue has been fixed. We support fuzzy-search for the boards:

[image]
[image]
[image]
[image]
[image]
[image]
----

Regarding the second part of the issue and the `arduino-esp8266 is not a valid FQBN` error; it is tracked already [here](https://github.com/arduino/arduino-pro-ide/issues/272). The problem is caused by the hyphen (`‐`) in the FQBN. I have already removed the flawed FQBN validation and we rely on the CLI from the Pro IDE.

I consider this task as done and will be available in the next release.

ubidefeo, 2020-07-20:
yes!
thank you, @kittaakos

kittaakos, 2020-07-21:
> I consider this task as done

I  do not like the fuzzy search as-is; I am going to rework the boosting a bit.

closed 2020-08-26 by ubidefeo