arduino/arduino-pro-ide #264: PRO-IDE Alpha 0.0.6

lamidou, 2020-06-04:
I've started using the "pro-ide" and encountered some problems (which is not  a surprise = alpha)
This is an EXCELLENT piece of software, clear, efficient... The best idea since the arduino itself !

I have a problem with forward declarations.

**CASE A** : your function() is above the function LOOP() so you don't need any declaration because your function() is before the code where you call it. This works OK.

**CASE B** : If you have plenty of or lengthy functions, you rather place them at the end of the code (that is after the main section (setup() and loop() ) so you can reach quickly the main section. This is convenient but all declarative languages need a so called forward declaration of these functions in order to know about at compile time.
The arduino IDE takes care of that and you get no error when omitting these forward declarations. This works OK with both IDE (standard and Pro)
BUT the pro IDE underlines (red wave) all calls to these functions because C needs a forward declaration. The compilation works OK without errors. But reading the code is very messy to read. So let's try the next case.

**CASE C** : In order to prevent the problems seen in case A and B you just need to add these forward declaration before any call to these functions. This works OK, no error, no red waves, compiles as well.

_SO WTF ?_
 
There is a very useful feature : on the right of the proIDE you can open a panel called 'OUTLINE' you can toggle it using the icon under the blue ones on the top right.
This panel contains all identifiers used in you code, and when you click one, you jump directly to it... and this is great !
BUT when you have forward declarations you jump to it and not to the function() itself which is a real pain...

**SUGGESTION** : keep the forward declaration principle because that is the right way to think in a declarative language, but get the feature 'OUTLINE' to jump to the function() and not to forward declaration.

Thanks !

jackthese, 2020-06-11:
Frankly, I see no value in requiring forward declarations in C/C++, just duplicative typing and excessive clutter. Compilers and IDEs should facilitate clarity and ease of use, and not be chained to an archaic concept that should be handled automatically and invisibly. The warnings should be suppressed.

Dave528, 2020-06-17:
> Frankly, I see no value in requiring forward declarations in C/C++, just duplicative typing and excessive clutter. Compilers and IDEs should facilitate clarity and ease of use, and not be chained to [...]

Honestly, while Arduino is not truly a C/C++ environment, I agree that since the forward declaration is allowed (and obviously used, if only for syntax highlighting) the outliner should be bringing the code view to the definition of the jump and NOT the forward declaration.  Might even be nice to have a context menu jump selection to go to either the declaration or definition, with a jump to the definition being the default.

per1234, 2021-02-17:
The language server now understands Arduino's automatically generated forward declarations system in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4), so there is no longer a need to add them manually to avoid spurious red underlines. The outline system is now working for me as expected even when I do manually add forward declarations.

closed 2021-02-17 by per1234