arduino/arduino-pro-ide #267: Mac OS cannot complete verification of "Arduino Pro IDE" at launch

toddewilliams, 2020-06-09:
**Describe the bug**
Upon launch of the 0.0.6 Alpha release my Mac shows the normal "Verifying Arduino Pro IDE" status dialog. However, the status never advances and the CoreServicesUIAgent task eventually consumes so much RAM that the machine is unusable and the process must be forcibly killed.

**To Reproduce**
Steps to reproduce the behavior:
1. Download 'arduino-pro-ide-0.0.6-alpha.preview-osx.dmg'
2. Launch the dmg and drag the Arduino Pro IDE into the Applications folder
3. Launch "Arduino Pro IDE"
4. Open "Activity Monitor" and note that RAM consistently increases for the CoreServicesUIAgent until the process must be forcibly killed. 

**Expected behavior**
I expected the OS to quickly verify the application and launch it.

**Screenshots**
When I finally managed to kill it the first time it had consumed 11GB. I took this screenshot on a second run, but before it locked up the machine.

[image]

**Desktop (please complete the following information):**
 - OS: MacOS El Capitan
 - Version: 10.11.6

kittaakos, 2020-06-11:
> Upon launch of the 0.0.6 Alpha release my Mac shows the normal "Verifying Arduino Pro IDE" status dialog.

Let's see if I understood this, so you got the dmg, you could unpack it to `/Applications/` and when you start it hangs. Is that correct?

Few questions:
 - Do you have the `arduino-cli` somewhere on your `$PATH`?
 - What do you see if you start the app from a terminal? (`/Applications/Arduino\ Pro\ IDE.app/Contents/MacOS/Arduino\ Pro\ IDE`)

toddewilliams, 2020-06-11:
> Let's see if I understood this, so you got the dmg, you could unpack it to /Applications/ and when you start it hangs. Is that correct?

Yes. That is correct.

>Do you have the arduino-cli somewhere on your $PATH?

I do not, yet. I was in a rush to have a look at the IDE.

>What do you see if you start the app from a terminal? (/Applications/Arduino\ Pro\ IDE.app/Contents/MacOS/Arduino\ Pro\ IDE)

The IDE launches correctly from the terminal. I'll speculate that this because the terminal launch bypasses the OS verification that is being attempted with a standard double-click launch. In any event, this is a viable workaround to this issue.

I've attached the terminal output in case it is informative.

[Issue267TerminalOutput.txt](https://github.com/arduino/arduino-pro-ide/files/4765102/Issue267TerminalOutput.txt)

kittaakos, 2020-06-11:
Thanks for the update, @toddewilliams.

> I'll speculate that this because the terminal launch bypasses the OS verification that is being attempted with a standard double-click launch

I've never seen this or a similar issue before. We know, the macOS app has to be notarized (#5), but it should not hang at app startup.

toddewilliams, 2020-06-11:
This could be a local issue since I'm running a rather old version of MacOS, for "reasons." Anyway, if it doesn't affect anyone on a recent MacOS release and since launching from the terminal is a viable workaround then please don't waste time looking into this further. I've got the ability to launch and that's what I needed. 

I'm not an Arduino developer. I simply wanted to try installing our pair programming solution from CodeTogether.com into the IDE since it's based on VS Code in the hopes that a free pair programming solution would be of use to your community.

kittaakos, 2021-02-11:
This is available, the new IDE is signed. Let us know if you still have the same issue. Closing.

closed 2021-02-11 by kittaakos