arduino/arduino-pro-ide #270: Restrict the app to one instance

kittaakos, 2020-06-11:
Due to this: https://github.com/eclipse-theia/theia/issues/7983

Users will be still able to open multiple browser windows (per sketches) for the same process, but they will **NOT** be able to start the Pro IDE twice.

ubidefeo, 2021-02-18:
The main reason (probably only) to run two instances of the Java IDE was a hack which allowed users to have 2 serial monitors open at once.
In the new IDE each window (sketch) can have its own Serial Monitor
[image]

@per1234 do you think it's fair to close this as "won't do"?

per1234, 2021-02-18:
If I understand https://github.com/eclipse-theia/theia/issues/7983 correctly (unlikely), Akos's proposal is intended to avoid a bad experience for the user by preventing them from doing something that is not supported (starting two instances of Arduino Pro IDE). That sounds like a very good thing to me.

Even if it's not considered a high priority to work on right now, I would recommend leaving it open for a later time when the higher priority work is finished, or if we find that https://github.com/eclipse-theia/theia/issues/7983 is a regular source of problems for the users.

ubidefeo, 2021-02-18:
@per1234 my point was mostly about reasons for opening multiple instances of the IDE.
Since multiple windows/sketches can use multiple Serial Monitor instances, a very common reason for doing so is to have multiple serial monitors connected to multiple boards.
2 instances of this IDE would also conflict because of the gRPC port binding to the CLI.

I cannot see any reason not to restrict it

per1234, 2021-02-18:
>I cannot see any reason not to restrict it

Same here, so Akos's proposal to "Restrict the app to one instance" is reasonable. So, unless this improvement has already been done, I would leave this open.