arduino/arduino-pro-ide #271: how to use net proxy?

Y-A-K-E, 2020-06-11:


kittaakos, 2020-06-18:
@Y-A-K-E, I am pretty sure there are limitations for proxies. Please explain your use-case a bit more details. Do you need it for the 3rd party cores? Thanks!

Y-A-K-E, 2020-06-19:
China can't open github.com . we need to set a proxy server to download GitHub lib.

kittaakos, 2020-06-19:
> China can't open github.com .

Understood. 👍 

> we need to set a proxy server to download GitHub lib.

What do you want to download from a GitHub lib? Do you want to pull 3rd part cores? If so, I think this should be configured in the CLI config. Is it supported, @cmaglie?

cmaglie, 2020-06-22:
yes, the support has been added in arduino-cli 0.10.0 (https://github.com/arduino/arduino-cli/pull/609) you must add in the configuration file the keyword:

`network.proxy`

for example:

`network.proxy=http://username:password@1.2.3.4:8883/`

kittaakos, 2020-06-22:
> yes, the support has been added in arduino-cli 0.10.0

Super. Thank you!

per1234, 2021-02-17:
Proxy can now be configured via the "Network" tab of Arduino Pro IDE's  **File > Preferences** dialog, so I'm going to close this as fixed. @Y-A-K-E if you try out [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4) and find that it doesn't meet your needs, please comment here and we'll investigate.

closed 2021-02-17 by per1234