arduino/arduino-pro-ide #272: NodeMCU-32S not recognized as valid board

jackthese, 2020-06-11:
**Describe the bug**
When NodeMCU-32S board is selected and an upload is requested, the error message is "Error: esp32:esp32:nodemcu-32s is not a valid FQBN."

**To Reproduce**
Steps to reproduce the behavior:
1.Go to board manager and select subject board, which is highlighted properly as a valid board
2. Cntl>Alt>U to attempt uploading.

**Expected behavior**
Nominal compile followed by upload (works OK in standard IDE)

**Screenshots**
Don't know how to do that easily, but basically above error message is a notification.

**Desktop (please complete the following information):**
 - OS: Windows 10 
 - Version 1909

**Additional context**
Add any other context about the problem here.

per1234, 2020-06-22:
Hi @jackthese. Thanks so much for this helpful report! The reason you got this error is because the Pro IDE doesn't currently support FQBNs (fully qualified board name) that contain `-`. The board ID of your NodeMCU-32S happens to contain this character.

The Arduino developers are aware of this and working to add support to the Pro IDE for hyphenated FQBNs.

Until then, I can provide you with a workaround:

1. (In the Arduino Pro IDE) **File > Settings > Open CLI Configuration**
2. Note the data directory path that is defined in the `directories.data` field (It will likely be `C:\Users\{username}\AppData\Local\Arduino15`). You don't need to make any changes to the arduino-cli.yml file.
3. Open the data directory. Note that `C:\Users\{username}\AppData` is a system folder, and thus hidden by default. You will need to configure Windows File Explorer to show hidden items before you can see it.
4. Open the `packages/esp32/hardware/esp32/1.0.4` subdirectory of the data directory.
5. Open the file `boards.txt` in a text editor.
6. Change all instances of `nodemcu-32s.` in the file to `nodemcu32s.`
7. Save the file.
8. Close all Arduino PRO IDE windows, then start the Pro IDE again.

You should now be able to use your NodeMCU-32S board in the Pro IDE without getting that "is not a valid FQBN" error (because you removed the `-` from the board ID, and thus changed the FQBN to `esp32:esp32:nodemcu32`). This change has no other impact. The board ID is merely an arbitrary name used to identify the board and any unique name works as well as the other.

jackthese, 2020-06-22:
Perfect answer, thanks. Note that I had to drill down to a slightly different path (has /Local in it). 

C:\Users\jackr\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4

 

Now, if you had a workaround to enable opening multiple files in the editor, like VS Code has, I would make Pro my go-to IDE for Arduino work. Thanks again for you quick reply….Jack

GhaithKhmili, 2020-07-13:
Hello,
I am having the same issue with ESP32 DEVKIT V1
Error: esp32:esp32:esp32doit-devkit-v1 is not a valid FQBN.

per1234, 2020-07-13:
Hi @GhaithKhmili you can solve this by following the same instructions I posted in my previous reply (https://github.com/arduino/arduino-pro-ide/issues/272#issuecomment-647436357).

The only difference is that when you get to step 6:
>6. Change all instances of `nodemcu-32s`. in the file to `nodemcu32s`.

You should instead change all instances of `esp32doit-devkit-v1` in the file to `esp32doitdevkitv1`.

GhaithKhmili, 2020-07-13:
> 
> 
> Hi @GhaithKhmili you can solve this by following the same instructions I posted in my previous reply ([#272 (comment)](https://github.com/arduino/arduino-pro-ide/issues/272#issuecomment-647436357)).
> 
> The only difference is that when you get to step 6:
> 
> > 1. Change all instances of `nodemcu-32s`. in the file to `nodemcu32s`.
> 
> You should instead change all instances of `esp32doit-devkit-v1` in the file to `esp32doitdevkitv1`.

Hello @per1234,
Thank you for your quick response. I actually removed all the hyphens as indicated in the provided solution yet I am still getting the same error.

per1234, 2020-07-13:
Did you restart the Arduino Pro IDE (step 10)? This must be done before any edits to boards.txt will take effect.

GhaithKhmili, 2020-07-13:
> 
> 
> Did you restart the Arduino Pro IDE (step 10)? This must be done before any edits to boards.txt will take effect.

Yes I did save the file and restart the IDE with every attempt.

kittaakos, 2020-07-13:
> Yes I did save the file and restart the IDE with every attempt.

Do you have another board, @GhaithKhmili? Try to select that other board, then the (updated) `esp32doitdevkitv1` again. Does it help?

GhaithKhmili, 2020-07-14:
Hello @kittaakos, 
I don't have another board but your suggestion gave me the idea to go to the board selector, reselect both board and COM port once again and that actually worked. Thank you.

kittaakos, 2020-07-14:
> reselect both board and COM port once again and that actually worked.

Thank you for the update, I am glad it worked. Smart, yes, that should have the same effect. 👍 

Some background: once you have selected a board that was recognized by the CLI, the app caches the FQBN of the board. So even if you updated the FQBN outside of the app, the app did not know anything about this change.

per1234, 2021-02-17:
Support for hyphenated FQBN was added in Arduino Pro IDE 0.1.0 last August but we forgot to close this. So the workarounds to use the ESP32 and other boards described above are no longer necessary

closed 2021-02-17 by per1234