arduino/arduino-pro-ide #273: Is this project/forum dead?

jackthese, 2020-06-13:
I see bug reports from as early as Oct. 19, 2019 that have received no followup comments from the development team, suggesting to me that this project is dead. Hope that's wrong, because there are some nice features I would like to use. Anyone out there to comment?

per1234, 2020-06-13:
The [last release](https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.6-alpha.preview) (which fixed many bugs and provided much requested features) was made only a little over a month ago.

The developers regularly respond and act on the issues reported here.

So clearly this project is not dead.

If you look at the commit history of the repository, it does give the impression of inactivity, but this is misleading. That is because this repository is currently only used to store releases, some documentation, and the issue tracker. The development work is happening in a separate repository.

Sure, there are open issues, I think you'll find the same in any large project.

jackthese, 2020-06-22:
OK, good to hear. One “Enhancement” I’ve posted on your repository is the fact that I can’t view multiple files within my project. The right click dropdown menus for opening files do not show the same as VS Code. I tend to divide my sketches into functional files, each with a header, and looking at that structure one file at a time is a major shortcoming of the regular and Pro Arduino IDSs for me. Has forced me over to VS Code, and all the problems associated with both the vanilla Arduino and Platform IO implementations. My other post on your repository (as a bug) is that the ESP32 board NodeMCU-32s is listed as a valid board in your board manager, but does not work. See my post for details.

 

You’re right, the lack of even an acknowledgement of my posts led me to believe that nothing was ever going to be done about these issues, and that the effort was dead. I suspect many others have come to the same conclusion, because there are few repeats from them, suggesting that they have stopped trying to help the project be a success.

 

Is there a better place to actually get some interaction on things like this? Frankly, if this is as good as it gets, I’ll work to make the VS Code implementations better by using them from time to time…Jack

per1234, 2020-06-22:
I really appreciate you taking the time to make high quality feature requests and bug reports. These really are valuable contributions.

Since I think I have addressed the items you brought up specific to this issue, I'll go ahead and close this. I hope you will continue to watch this project to see the ongoing progress that is happening.

closed 2020-06-22 by per1234