arduino/arduino-pro-ide #274: Types defined in libraries not recognized by language server when sketch is outside user directory

per1234, 2020-06-14:
**Describe the bug**
When a sketch is located anywhere outside the user directory (AKA "sketchbook"), the language server reports types defined in libraries used by the sketch as unknown.

**To Reproduce**
Steps to reproduce the behavior:
1. Install the Servo library (any library that defines a type will work).
1. Open the "Sweep" example sketch that comes with the Servo library (any sketch using a type defined in a library will work). Note that the `Servo` type is accepted by the language server:
[image]
1. Copy the "Sweep" sketch to somewhere outside the user directory.
1. Open the copy of the "Sweep" sketch. Note that the `Servo` type is now not recognized by the language server:
[image]

**Expected behavior**
The language server to recognize library defined types regardless of the location of the sketch.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04, Windows 10
 - Version: 0.0.6

**Additional context**
Originally reported at: https://forum.arduino.cc/index.php?topic=690091

StarMiner99, 2020-06-15:
Same Issue

StarMiner99, 2020-06-15:
could not enable editor support was something with arduino-cli i removed it from my path so it uses th e built-in one now it works fine

yonatan8070, 2020-07-19:
I have the same issue, but when the new .ino file is still within the same folder, like this:
```
Some_Folder
L Some_Folder.ino (type from library gets recognized)
L Another_File.ino (type from library shows as error)
```

RussellAult, 2020-10-16:
I'm running into this problem even for sketches within the User directory.

ubidefeo, 2020-10-17:
@RussellAult 
see my reply on #65

novakkry, 2020-11-13:
> I have the same issue, but when the new .ino file is still within the same folder, like this:
> 
> ```
> Some_Folder
> L Some_Folder.ino (type from library gets recognized)
> L Another_File.ino (type from library shows as error)
> ```

Have you solved the problem? I'm having the same problem - the IDE underlines the 'problems' in "Another_File.ino". In the "Some_folder.ino" there is no underlining. It compiles fine.

per1234, 2021-02-17:
Great news everyone! This is fixed in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4).

closed 2021-02-17 by per1234